hyperkineticnerd/OpenCorsairLink #7: I'm wondering something here.

0x70ad57001, 2015-03-07:
Currently im porting this to C and to be a shared library, if you must know why im porting down to C.
But i have a question for you. 

A) Once this port is done would you accept it as a pull request? 
B) Do you mind if i changed the license to LGPL for the library? 

Thank you.

audiohacked, 2015-03-08:
I will accept a pull request, but can you do a full conversion of the library and have 1 or 2 source files for a C++ interface? As for the license, it was intended that the library was LGPL. I'm actually moving the library to its own git repo and set it as a submodule of this project.

0x70ad57001, 2015-03-08:
That wouldn't be a problem at all, ill make it so the API is very simplistic. Shortly i will create a repo and upload the current progress so you can see the progress of it so you tell me what you think of it.

0x70ad57001, 2015-03-12:
Sometime today i will be pushing the concept. Please look through and report back to what you may like or dislike. The repo will be here; https://github.com/0x70ad57001/libcorsairlink

audiohacked, 2015-03-12:
Code is looking good, but you need to add "Lesser" to any strings like GNU General Public License in accordance to LGPL licensing practices.

0x70ad57001, 2015-03-12:
That won't be an issue. As for getting it going got a few more things to-do before i write the make file and compile the first shared library object. after it compiles testing will begin.

0x70ad57001, 2015-03-30:
Currently fixing all compiler issues, and writing a set of unit-tests. In short the tests will make sure that libcorsairlink abides the same as OpenCorsairLink, in terms of current modes, current fan count current colors ect. Once it compiles and the tests are completed and passing, I do say the port is complete! :+1:

closed 2018-05-30 by audiohacked