hyperkineticnerd/OpenCorsairLink #8: Error building

towheed, 2015-04-13:
Just trying to build but fails with:
------
/usr/bin/ld: cannot find -lGL
collect2: error: ld returned 1 exit status
Makefile:159: recipe for target '../OpenCorsairLinkGui' failed
make[1]: *** [../OpenCorsairLinkGui] Error 1
make[1]: Leaving directory '/home/towheed/OpenCorsairLink/Gui'
Makefile:126: recipe for target 'sub-Gui-make_first-ordered' failed
make: *** [sub-Gui-make_first-ordered] Error 2
----------

Any assistance welcomed.

Thanks for supporting this on Linux.  Hope I can get it working.

Towheed

audiohacked, 2015-04-13:
What Distribution is this on? It looks like its trying to link the OpenGL library, even though there's no dependency on OpenGL.

towheed, 2015-04-13:
I'm on Debian Testing (Jessie).  I was thinking the same and verified I had the OpenGL libs installed.

It's built to the point where I can run the CLI.  I've only just tested using the -p and -i switches but would also like to get the GUI running.

towheed, 2015-04-14:
Here's the line in the Gui/Makefile:
LIBS          = $(SUBLIBS) -L/usr/X11R6/lib64 -L.. -lCorsairLink -lhidapi-hidraw -lQt5Widgets -lQt5Gui -lQt5Core -lGL -lpthread

GL?  What lib is that? OpenGL?

towheed, 2015-04-14:
Removed -lGL from the Makefile and got the it to complete the build but it's very slow to respond to clicks on the menu items.  Any way to set fan speed from the GUI?

audiohacked, 2015-04-16:
Maybe make sure Qt5 is built with OpenGL!? There's a reason why I don't develop on debian, ubuntu, or fedora...

towheed, 2015-04-16:
I think it is....but the Qt5 OpenGL dev libs are also installed.

This is a new build, I'll see if it has anything to with the nVidia drivers as I installed directly and not from the Debian packages.

The CLI works wonderful, thanks for that.
Is there some way to interface this with say Conky, just as an example, or even lm-sensors.

closed 2018-05-30 by audiohacked