hyperkineticnerd/OpenCorsairLink #9: Synology chroot synagios

stvnmbr1, 2015-08-17:
Hi,

I have a syno with chroot installed and the link isn't working.. I tried everything I could find online, but no sigar :( it's picking up something on the controllers, OpenCorsairLink still says cannot initialize link.. 
i used this tutorial https://forum.teksyndicate.com/t/corsair-h100i-linux-howto/75829

any ideas?

if I use
sudo bash -c "echo -n 3-2:1.0 >/sys/bus/usb/drivers/usbhid/unbind"

lsusb sees 

lsusb -d 1b1c:0c04 -v
 
Bus 003 Device 002: ID 1b1c:0c04 Corsair
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c04
  bcdDevice            2.00
  iManufacturer           1 Corsair Memory, Inc.
  iProduct                2 Integrated USB Bridge
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      37
          Report Descriptor: (length is 37)
            Item(Global): Usage Page, data= [ 0x00 0xff ] 65280
                            (null)
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x40 ] 64
            Item(Local ): Usage Minimum, data= [ 0x01 ] 1
                            (null)
            Item(Local ): Usage Maximum, data= [ 0x40 ] 64
                            (null)
            Item(Main  ): Input, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x40 ] 64
            Item(Local ): Usage Minimum, data= [ 0x01 ] 1
                            (null)
            Item(Local ): Usage Maximum, data= [ 0x40 ] 64
                            (null)
            Item(Main  ): Output, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0080
  (Bus Powered)

with
sudo bash -c "echo -n 3-2:1.0 >/sys/bus/usb/drivers/usbhid/bind"

HID Device Descriptor is unavailable

other outputs:

root@StevieSyn:/# cat /sys/class/usbmisc/hiddev9/uevent
MAJOR=180
MINOR=105
DEVNAME=usb/hiddev9
PHYSDEVPATH=/devices/pci0000:00/0000:00:1d.2/usb3/3-2/3-2:1.0
PHYSDEVBUS=usb
PHYSDEVDRIVER=usbhid

root@StevieSyn:~/hidapi/OpenCorsairLink# sudo ./OpenCorsairLinkCli -
i
sudo: unable to resolve host StevieSyn
Error: Unable to open Corsair H80i, H100i or H110i CPU Cooler
Cannot initialize link.

root@StevieSyn:~/hidapi/OpenCorsairLink# cat /sys/class/usbmisc/hidd
ev9/device/uevent
DEVNAME=3-2:1.0
DEVTYPE=usb_interface
DRIVER=usbhid
PHYSDEVBUS=usb
PHYSDEVDRIVER=usbhid
PRODUCT=1b1c/c04/200
TYPE=0/0/0
INTERFACE=3/0/0
MODALIAS=usb:v1B1Cp0C04d0200dc00dsc00dp00ic03isc00ip00in00

root@StevieSyn:~/hidapi/OpenCorsairLink# cat /sys/class/usbmisc/hidd
ev9/device/0003:1B1C:0C04.0002/uevent
DEVNAME=0003:1B1C:0C04.0002
DRIVER=hid-generic
PHYSDEVBUS=hid
PHYSDEVDRIVER=hid-generic
HID_ID=0003:00001B1C:00000C04
HID_NAME=Corsair Memory, Inc. Integrated USB Bridge
HID_PHYS=usb-0000:00:1d.2-2/input0
HID_UNIQ=
MODALIAS=hid:b0003g0001v00001B1Cp00000C04

root@StevieSyn:~/hidapi/OpenCorsairLink# cat /sys/class/usbmisc/hidd
ev9/device/0003:1B1C:0C04.0002/report_descriptor
ÿ      ¡&ÿ•@)@   •@)@‘       Àroot@StevieSyn:~/hidapi/Op
ev9/device/0003:1B1C:0C04.0002/report_descriptor
ÿ      ¡&ÿ•@)@   •@)@‘       Àroot@StevieSyn:~/hidapi/Op
inCorsairLink# sudo ./OpenCorsairLinkCli -
sudo: unable to resolve host StevieSyn
Error: Unable to open Corsair H80i, H100i or H110i CPU Cooler
Cannot initialize link.

root@StevieSyn:~/hidapi/OpenCorsairLink# cat /sys/class/usbmisc/hidd
ev9/device/uevent
DEVNAME=3-2:1.0
DEVTYPE=usb_interface
DRIVER=usbhid
PHYSDEVBUS=usb
PRODUCT=1b1c/c04/200
TYPE=0/0/0
INTERFACE=3/0/0
MODALIAS=usb:v1B1Cp0C04d0200dc00dsc00dp00ic03isc00ip00in00

root@StevieSyn:~/hidapi/OpenCorsairLink# cat /sys/class/usbmisc/hidd
ev9/uevent
MAJOR=180
MINOR=105
DEVNAME=usb/hiddev9
PHYSDEVPATH=/devices/pci0000:00/0000:00:1d.2/usb3/3-2/3-2:1.0
PHYSDEVBUS=usb
PHYSDEVDRIVER=usbhid
root@StevieSyn:~/hidapi/OpenCorsairLink# usbhid-dump

lsusb
Bus 002 Device 002: ID 1130:cc00 Tenx Technology, Inc.
Bus 003 Device 002: ID 1b1c:0c04 Corsair

~/hidapi/OpenCorsairLink# usbhid-dump -d 1b1c:0c04
003:002:000:DESCRIPTOR         1439828913.087843
 06 00 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 19 01 29 40 81 02 09 01 75 08 95 40 19 01 29 40
 91 02 09 01 C0

stvnmbr1, 2015-08-17:
and also tried https://github.com/audiohacked/OpenCorsairLink/wiki/How-to-build-and-use., still not working..

stvnmbr1, 2015-08-17:
i got a little further..

lsusb -d 1b1c:0c04 -v
 
Bus 003 Device 002: ID 1b1c:0c04 Corsair
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c04
  bcdDevice            2.00
  iManufacturer           1 Corsair Memory, Inc.
  iProduct                2 Integrated USB Bridge
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      37
          Report Descriptor: (length is 37)
            Item(Global): Usage Page, data= [ 0x00 0xff ] 65280
                            (null)
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x40 ] 64
            Item(Local ): Usage Minimum, data= [ 0x01 ] 1
                            (null)
            Item(Local ): Usage Maximum, data= [ 0x40 ] 64
                            (null)
            Item(Main  ): Input, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x40 ] 64
[... 128 lines cut ...]
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      37
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0080
  (Bus Powered)
root@StevieSyn:/#
root@StevieSyn:/# cd ~/hidapi/OpenCorsairLink
root@StevieSyn:~/hidapi/OpenCorsairLink# sudo ./OpenCorsairLinkCli -i
sudo: unable to resolve host StevieSyn
Error: Unable to open Corsair H80i, H100i or H110i CPU Cooler
Cannot initialize link.

audiohacked, 2015-08-18:
I don't know whats going on here, but it looks like the Synology chroot is really messing with the hardware access. Also, sudo is not necessary while running as root.

stvnmbr1, 2015-08-18:
I thought i was some permission issue but I don't know for sure.. 

there's like a double udev, one for syno and one for chroot in syno..

and by saying that, could it be that because it's binded in syno, chroot can't access it fully.. but then if i would unbind him from syno, would chroot still see it? maybe it's something else..

i have a udev in /lib from syno root with a rules.d ( i think this is the issue, but i don't have enough experience with this..) and lots of files, i'll post a dir output in a while

and one in /var/packages/Synagios/target/nagios-chroot/root/hidapi/udev

and one in /var/packages/Synagios/target/nagios-chroot/lib/udev with also a rules.d full of files

stvnmbr1, 2015-08-18:
SYNO ROOT 

/lib/udev/rules.d$ dir
drwxr-xr-x    2 root     root          4096 Aug 18 10:04 .
drwxr-xr-x    6 root     root          4096 Aug 18 10:04 ..
-rw-r--r--    1 root     root            62 Aug 18 10:04 05-system-default.rules
-rw-r--r--    1 root     root           413 Jun 29 12:46 05-system-env.rules
-rw-r--r--    1 root     root           310 Jun 29 12:46 40-check-rdx.rules
-rw-r--r--    1 root     root           354 Jun 29 12:46 40-usb-env.rules
-rw-r--r--    1 root     root           448 Jun 29 12:46 45-update-disk-info.rules
-rw-r--r--    1 root     root           668 Jun 29 12:46 50-disk.rules
-rwxr-xr-x    1 root     root           243 Jun 29 12:46 50-firmware.rules
-rwxr-xr-x    1 root     root           325 Jun 29 12:46 50-hiddev.rules
-rwxr-xr-x    1 root     root           315 Jun 29 12:45 50-iscsi.rules
-rw-r--r--    1 root     root           168 Jun 29 12:46 50-net.rules
-rw-r--r--    1 root     root           427 Jun 29 12:46 50-printer-bus.rules
-rw-r--r--    1 root     root          1609 Jun 29 12:46 50-printers.rules
-rwxr-xr-x    1 root     root           406 Jun 29 12:46 50-scsi.rules
-rw-r--r--    1 root     root           360 Jun 29 12:46 50-scsi_host.rules
-rw-r--r--    1 root     root           337 Jun 29 12:44 50-tty.rules
-rwxr-xr-x    1 root     root          1014 Jun 29 12:46 50-ups.rules
-rw-r--r--    1 root     root           601 Jun 29 12:44 50-usb-bluetooth.rules
-rwxr-xr-x    1 root     root          1018 Jun 29 12:46 50-usb-dvb.rules
-rw-r--r--    1 root     root           423 Jun 29 12:44 50-usb-modem.rules
-rw-r--r--    1 root     root           448 Jun 29 12:44 50-usb-wifi.rules
-rwxr-xr-x    1 root     root           484 Jun 29 12:46 50-usb.rules

SYNO NAGIOS CHROOT

/var/packages/Synagios/target/nagios-chroot/root/hidapi/udev$ dir
drwxr-xr-x    2 root     root          4096 Aug 17 20:33 .
drwxr-xr-x   17 root     root          4096 Aug 17 20:34 ..
-rw-r--r--    1 root     root          1524 Aug 17 20:33 99-hid.rules

/var/packages/Synagios/target/nagios-chroot/lib/udev$ dir
drwxr-xr-x    5 root     root          4096 Aug 17 20:49 .
drwxr-xr-x   12 root     root          4096 Aug 17 20:49 ..
-rwxr-xr-x    1 root     root         38440 Apr 28  2013 accelerometer
-rwxr-xr-x    1 root     root         21932 Apr 28  2013 ata_id
-rwxr-xr-x    1 root     root         30136 Apr 28  2013 cdrom_id
-rwxr-xr-x    1 root     root         17932 Apr 28  2013 collect
-rwxr-xr-x    1 root     root           616 Apr 28  2013 create_static_nodes
drwxr-xr-x    2 root     root          4096 Apr 28  2013 devices
-rwxr-xr-x    1 root     root          2284 Apr 28  2013 dsl-modem.agent
-rwxr-xr-x    1 root     root         17848 Apr 28  2013 edd_id
-rwxr-xr-x    1 root     root          2314 Apr 28  2013 findkeyboards
-rwxr-xr-x    1 root     root           813 Apr 28  2013 firmware.agent
-rw-r--r--    1 root     root          1175 Apr 28  2013 hotplug.functions
-rwxr-xr-x    1 root     root           564 Dec  9  2012 hwclock-set
-rwxr-xr-x    1 root     root           502 Apr 28  2013 keyboard-force-release.sh
-rwxr-xr-x    1 root     root         30084 Apr 28  2013 keymap
drwxr-xr-x    3 root     root          4096 Jan  9  2014 keymaps
-rwxr-xr-x    1 root     root           614 Apr 28  2013 logger.agent
-rwxr-xr-x    1 root     root          5456 Apr 28  2013 mtd_probe
-rwxr-xr-x    1 root     root          2108 Apr 28  2013 net.agent
-rwxr-xr-x    1 root     root         26056 Apr 28  2013 pci-db
-rw-r--r--    1 root     root          2944 Apr 28  2013 rule_generator.functions
drwxr-xr-x    2 root     root          4096 Aug 17 20:49 rules.d
-rwxr-xr-x    1 root     root         30772 Apr 28  2013 scsi_id
-rwxr-xr-x    1 root     root         38528 Apr 28  2013 udev-acl
-rwxr-xr-x    1 root     root          7060 Mar 10  2014 udisks-dm-export
-rwxr-xr-x    1 root     root          7324 Mar 10  2014 udisks-lvm-pv-export
-rwxr-xr-x    1 root     root         30648 Mar 10  2014 udisks-part-id
-rwxr-xr-x    1 root     root          4260 Mar 10  2014 udisks-probe-ata-smart
-rwxr-xr-x    1 root     root          4648 Mar 10  2014 udisks-probe-sas-expander
-rwxr-xr-x    1 root     root         26056 Apr 28  2013 usb-db
-rwxr-xr-x    1 root     root          5448 Apr 28  2013 v4l_id
-rwxr-xr-x    1 root     root          3134 Apr 28  2013 write_cd_rules
-rwxr-xr-x    1 root     root           599 Apr 28  2013 write_dev_root_rule
-rwxr-xr-x    1 root     root          3936 Apr 28  2013 write_net_rules
[... 13 lines cut ...]
-rw-r--r--    1 root     root          2400 Apr 28  2013 60-persistent-input.rules
-rw-r--r--    1 root     root           947 Apr 28  2013 60-persistent-serial.rules
-rw-r--r--    1 root     root           912 Aug 27  2011 60-persistent-storage-dm.rules
-rw-r--r--    1 root     root          1464 Apr 28  2013 60-persistent-storage-tape.rules
-rw-r--r--    1 root     root          6598 Apr 28  2013 60-persistent-storage.rules
-rw-r--r--    1 root     root           770 Apr 28  2013 60-persistent-v4l.rules
-rw-r--r--    1 root     root           166 Apr 28  2013 61-accelerometer.rules
-rw-r--r--    1 root     root          4915 Jun  1  2012 69-cd-sensors.rules
-rw-r--r--    1 root     root          2569 Apr 28  2013 70-udev-acl.rules
-rw-r--r--    1 root     root           452 Apr 28  2013 75-cd-aliases-generator.rules
-rw-r--r--    1 root     root           642 Apr 28  2013 75-net-description.rules
[... cut ...]
-rw-r--r--    1 root     root            92 Jun 24  2012 85-hdparm.rules
-rw-r--r--    1 root     root           171 Dec  9  2012 85-hwclock.rules
-rw-r--r--    1 root     root          4372 Aug 17 12:14 91-permissions.rules
-rw-r--r--    1 root     root           847 Jun  1  2012 95-cd-devices.rules
-rw-r--r--    1 root     root          3320 Apr 28  2013 95-keyboard-force-release.rules
-rw-r--r--    1 root     root         11131 Apr 28  2013 95-keymap.rules
-rw-r--r--    1 root     root           155 Apr 28  2013 95-udev-late.rules

stvnmbr1, 2015-08-18:
Did the module update, still not working ..

I added

SUBSYSTEM=="usb", ATTR{idVendor}=="0x1b1c", ATTR{idProduct}=="0x0c04", MODE="0666"

to nagioschroot/lib/udev/rules.d/permissions.rules    is this the right permission rule? should i do this to all the rules.d ? sorry I'm like spamming this topic, but I have been trying to get the link to work for a whole week now :p

audiohacked, 2015-08-18:
I don't know anything about Synology or Nagios, so you will be on your own. Apparently your OS or Software is preventing anything else from working.

stvnmbr1, 2015-08-19:
damn.. i was happy you replied but wasn't the reply i'm looking for.. the nagios part i can figure out on my own.. 

can you give me some commands to test the output of the usb controller? like for testing the handle with hid_open. i could be the 0x that is messing with it, but if I change something in the code I probably have to execute the make command again.

but I really think it has to do something with the udev from syno (it's a linux kernel but there's no apt-get and such, therefore the debian chroot package. I read a lot about default rules and normally not everyone can access all info of the devices. that's why I thought you could help me with the udevs ?

would be really interesting for other xpenology users, no need for a VM to monitor and control the fans, Just plain Cli. You would make thousands of other users, very very happy if it could work :)

closed 2018-05-30 by audiohacked