hyperkineticnerd/OpenCorsairLink #12: Working on Support for H100i v2 and similar devices

audiohacked, 2016-09-07:
I've begun work on protocol work for the 'H100i v2', which I bought September 6th 2016. It might take sometime due to my class schedule and Senior Design project. No later than Spring 2017, I should have something working.

audiohacked, 2016-09-10:
Status update: Got some code to communicate with H100i v2!

TheOpenDevProject, 2016-10-26:
@audiohacked Do you have a patch for this? I would love to be able to control my H100i v2's LED (I know so lame)

TomAshley303, 2016-10-28:
I'm about to buy a v2 but I only use windows in a VM. I run linux as my main OS. Are my v2 fans going to be stuck at one speed if i don't have this software working? Shouldn't it throttle with the CPU temp?

TomAshley303, 2016-10-28:
nvm the fans will be plugged into my cpu fan socket I guess

TheOpenDevProject, 2016-10-29:
Although this is the complete wrong place, no the kernel knows how to control both the fans just fine. Just the LED cant be changed.

jontis, 2016-10-29:
I'm interested in helping. I have a Corsair RM650i connected with USB. My first goal is just to make it report energy consumption, but full Corsair Link functionality would be nice later on.

audiohacked, 2016-10-29:
Correction, the Fans and Pump are controlled by a CPU on the water cooler. The fan header is for power and reporting Fan speed. The motherboard doesn't control any Fan or Pump speeds! The Corsair Link software gives you the ability to change the fan curve, control LEDs, and Pump modes on the Water coolers.

matthewbaggett, 2016-11-07:
I also have a Corsair H100i v2.. I'd love to see your source! I was just messing around changing the PID to see if it would Just Work (tm)

audiohacked, 2016-11-08:
The Corsair H100i v2 uses a different company than the original H100i/H80i/H110i. Therefore, the protocol is different! I deeply apologize for not updating, but my school work is keeping me from my personal projects!

audiohacked, 2017-01-09:
There's a new version of OpenCorsairLink that supports the H115i, but it can currently only change LED colors. The old version of OpenCorsairLink has been moved to legacy branch.

sjug, 2017-02-23:
@audiohacked Have you had time to make progress? I'm considering a newer corsair, but it doesn't seem to be that useful in Linux.

audiohacked, 2017-02-23:
@sjug I've been pretty busy, and I'm still developing an API for supporting the old H100i, H100i v2, and RMi Power supplies. But I have been doing some work on the project lately!

jontis, 2017-02-24:
I'm interested in this project for supporting a PSU RM650i. Could you give some hints on how you extract the command flow and maybe I can do this work myself? At the moment, I'm running corsairs control software in Virtual Box.

audiohacked, 2017-02-26:
To extract the command flow: setup Wireshark with USB sniffing. This weekend I've had free time to work on RMi PSU support. The code is pretty hackish, but somewhat functional for monitoring.

jontis, 2017-02-27:
I started doing that a year ago but was very unsure of what to look for. 
Any way I can help you with simple tasks? I program Python for a living. Not that big on C.

Techwolf, 2017-02-28:
I have an H80i v2. I tried the legacy branch as that seem to be doing something while the master branch did nothing. Upon farther research, discovered the v2 has a different ID, 1b1c:0c08 while original H80i was 1b1c:0c04. I experimented with the legacy code and added the different ID. It still failed to find the H80iv2. Farther research reveled this was due to the kernel not creating the hidraw device for the H80i v2. More googling research, found out there is two different ways to talking to the HID USB devices. One was the hidraw and another was libusb that did not required a HID /dev entry. I got hopeful due to remembering the master branch required  libusb. So I tried the master branch again. Looking though the code I found where I could add the new ID. Using h80i got a libusb error. Then found h100i v2 and use that code with the h80i v2 ID. That worked, but upon running it for the first time, my fans went into jet takeoff mode. I had to reboot the computer to get things back to normal.

closed 2018-05-30 by audiohacked