hyperkineticnerd/OpenCorsairLink #22: Submiturb fails

GeoTob, 2017-03-20:
Just tried building from master on Ubuntu16.04 and ran into the error below. Wondering if you've ever come across something similar, or have any ideas how to fix?:)

```bash
$ sudo ./OpenCorsairLink.elf  --debug
usb device 9
corsair device 0
libusb: error [submit_control_transfer] submiturb failed error -1 errno=16
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
device_id 0x00
corsair device 1
libusb: error [submit_control_transfer] submiturb failed error -1 errno=16
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
device_id 0x00
corsair device 2
libusb: error [submit_control_transfer] submiturb failed error -1 errno=16
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
device_id 0x00
corsair device 3
libusb: error [submit_control_transfer] submiturb failed error -1 errno=16
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
device_id 0x00
corsair device 4
corsair device 5
corsair device 6
corsair device 7
```

`dmesg` shows a bunch of the following lines:
```bash
usbfs: process 23727 (OpenCorsairLink) did not claim interface 0 before use
```

And the `libusb-1.0` version in use is:
```bash
Source: libusb-1.0
Version: 2:1.0.20-1
```

audiohacked, 2017-03-21:
I've encountered the errors when I was developing the scan code to include detecting the device id for the older HID-based Hydro series. But on my test systems I've haven't seen that in over a week. In the testing branch, I have a patch to maybe fix the dmesg lines. Could you please try the testing branch?

audiohacked, 2017-03-21:
You also should be seeing device_id 0x00; the device_id used for the RMi and newer Hydro Coolers is 0xFF. You might want to be sure you're on the newest commit on master.

GeoTob, 2017-03-21:
Thanks for that, this has removed the libusb errors and dmesg messages, though device_id is now either 0x83 or 0x00, it changes "randomly" on re-running the elf ? 

Executable then exits with
```
DEBUG: scan done, start routines
DEBUG: device_number = -1
```

lsusb shows idProduct `0x0c04`

audiohacked, 2017-03-21:
To select your device you need to use '--device 0' if you have only one device. As for the device_id randomly changing, I don't know.

audiohacked, 2017-03-21:
What Corsair device do you have?

GeoTob, 2017-03-21:
Right, thanks for the fix. This segfaults now with `--device 0` (or any other device number), though I can try to find out myself if nothing immediately comes to your mind ?

```
[ 3408.275595] OpenCorsairLink[7256]: segfault at 18 ip 000000000040175e sp 00007ffc5613d420 error 4 in OpenCorsairLink.elf[400000+6000]
```

GeoTob, 2017-03-21:
It's a h110i // http://www.corsair.com/en-eu/hydro-series-h110i-280mm-extreme-performance-liquid-cpu-cooler

audiohacked, 2017-03-21:
The problem could be is your device isn't in the program database.

GeoTob, 2017-03-21:
I'll add it to the devices tonight and see how it goes

audiohacked, 2017-03-22:
If you can please post the output of "lsusb -d 1b1c: -v"?

GeoTob, 2017-03-22:
Full output below, didn't get very far adding it to the devices myself, apart from triggering a LED color change ;-)

```
Bus 001 Device 002: ID 1b1c:0c04 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c04 
  bcdDevice            2.00
  iManufacturer           1 Corsair Memory, Inc.
  iProduct                2 Integrated USB Bridge
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      37
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0080
  (Bus Powered)
```

closed 2018-05-30 by audiohacked