hyperkineticnerd/OpenCorsairLink #28: Cannot find file: OpenCorsairLink.pro

marcnotmark, 2017-06-18:
Hey all, sorry if this is a complete noob issue.  I'm relatively new, but i followed all the steps to install on the wiki and cloned the repository, but i'm getting an error when i try to do the "qmake OpenCorsairLink.pro" command.  Any help would be appreciated- thanks in advance!

Serubin, 2017-06-18:
@marcnotmark `OpenCorsairLink.pro` is provided in the [root of the repo](https://github.com/audiohacked/OpenCorsairLink/blob/legacy/OpenCorsairLink.pro).

Make sure it's present there. If it's now, download it, if it is can you run `ls -la .` in the root of the repo and show us the results?

marcnotmark, 2017-06-18:
Ah I see, let me try that now.  That looks to be on the legacy branch.. should i be pulling everything from there too?

audiohacked, 2017-06-18:
Please update to the latest release, I will not provide support on the legacy branch.

marcnotmark, 2017-06-19:
@audiohacked  I had initially downloaded the latest in the master branch but that does not have OpenCorsairLink.pro.

Serubin, 2017-06-19:
@marcnotmark In the new branch just type in `make` and it will create the elf

audiohacked, 2017-06-19:
@marcnotmark the master branch doesn't use qmake or OpenCorsairLink.pro. Also, the instructions are for the legacy branch.

closed 2018-05-30 by audiohacked