hyperkineticnerd/OpenCorsairLink #29: rm650i data

jontis, 2017-06-24:
I'm thinking as 850 works, this just needs a minor fix if someone would point me in the right direction. Output is now:
$ sudo ./OpenCorsairLink.elf --debug --device 0
usb device 1
corsair device 0
corsair device 1
corsair device 2
corsair device 3
corsair device 4
corsair device 5
corsair device 6
corsair device 7
device_id 0xFF
Dev=0, CorsairLink Device Found: RM650i!

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: 
Product: 
Firmware: NA
DEBUG: string done
03 FE 00 00 00 00
Temperature 0:  0.00 C
03 FE 00 00 00 00
Temperature 1:  0.00 C
Powered: 0 (0d.  0h)
Uptime: 0 (0d.  0h)
DEBUG: time done
03 FE 00 00 00 00
Supply Voltage:  0.00
03 FE 00 00 00 00
Total Watts:  0.00
DEBUG: supply done
Output 12v:
03 FE 00 00 00 00
	Voltage  0.00
03 FE 00 00 00 00
	Amps  0.00
03 FE 00 00 00 00
	Watts  0.00
Output 5v:
03 FE 00 00 00 00
	Voltage  0.00
03 FE 00 00 00 00
	Amps  0.00
03 FE 00 00 00 00
	Watts  0.00
Output 3.3v:
03 FE 00 00 00 00
	Voltage  0.00
03 FE 00 00 00 00
	Amps  0.00
03 FE 00 00 00 00
	Watts  0.00

jontis, 2017-06-25:
@fpoussin is it possible that your method could work here?

fpoussin, 2017-06-25:
Hey,

Actually, it seems that i'm still having the issue if I don't "initialize" the psu through corsair link using usb over IP on my desktop. (the psu is on a server)
Corsair link reads 64 bytes from the control endpoint at each command, which we don't, that might be worth trying.

This is the first few packets.
There are no special init commands as far as I can tell.
There's only one configuration and one interface so no tricks here.
[image]

jontis, 2017-06-25:
I think this outside my comfort zone... 
"In", is that going "in"to the PSU or "out" from the PSU?

fpoussin, 2017-06-25:
USB packet directions are always related to the host, so IN is a read from the PSU.

jontis, 2017-06-25:
I'm not used to this low level communication protocols.
Do you have a testbed where you can issue single commands and read responses without compiling C programs?

fpoussin, 2017-06-26:
You can do that with python:
https://gist.github.com/fpoussin/6c10f4a80e2792aed91bfb20c868dbca

jontis, 2017-07-02:
This was not my strong side:
~/code/openCorsairLink/py_usb $ sudo python3 py_usb.py 
Traceback (most recent call last):
  File "py_usb.py", line 60, in <module>
    dev.set_configuration()
  File "/usr/local/lib/python3.5/dist-packages/usb/core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/usr/local/lib/python3.5/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/usb/core.py", line 148, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "/usr/local/lib/python3.5/dist-packages/usb/backend/libusb1.py", line 794, in set_configuration
    _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
  File "/usr/local/lib/python3.5/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy

jontis, 2017-07-03:
Ok, so on with the work. That error was because some kernel driver had claimed the device.
This releases is:
if dev.is_kernel_driver_active(0):
    print('Kernel driver was active, detaching')
    dev.detach_kernel_driver(0)

jontis, 2017-07-03:
This is what I get out from it:
request_string(CLINK_GET_POWER)
Out[77]: '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

In [78]: request_string(CLINK_GET_PRODUCT)
Out[78]: '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

In [79]: request_string(CLINK_GET_INPUT)
Out[79]: '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

Can the instructions in ep_write be different for this PSU?

jontis, 2017-07-04:
Tinkered a bit, now running a virtualbox with windows and letting wireshark dump from usbmon.
These are the requests with answers so far:
(3, 154) = RM650i
(3, 216) = 2.0
(3, 142) = 42.0
(2, 0) = 0.0
(3, 0) = 0.0
(3, 150) =128.0

jontis, 2017-07-04:
Or in hex:
CLINK_GET_PRODUCT_650 = (0xFE, 0x03)
CLINK_GET_WHAT1 = (0x03, 0xD8)
CLINK_GET_WHAT2 = (0x03, 0x8E)
CLINK_GET_WHAT3 = (0x02, 0x00)
CLINK_GET_WHAT4 = (0x03, 0x00)
CLINK_GET_WHAT5 = (0x03, 0x96)

fpoussin, 2017-07-05:
Ah yes sometimes the device needs to be released.
I already have the commands, they are the same for all RMi PSUs.

jontis, 2017-07-05:
My bad, I interpreted the different commands in the guist as that there were differences in the RMIs api.
So all commands are already known?

fpoussin, 2017-07-06:
Maybe not all, but most of them: https://github.com/audiohacked/OpenCorsairLink/tree/master/protocol/rmi

jontis, 2017-07-06:
Ok, I was assuming something was missing as the program only reads 0 for everything.

fpoussin, 2017-07-10:
The commands are correct, but there's some init somewhere that we are missing

jontis, 2017-07-10:
anything I can test to help?

audiohacked, 2017-09-01:
Fix committed to master branch. Please test

closed 2018-05-30 by audiohacked