hyperkineticnerd/OpenCorsairLink #32: hx1000i

b0gd4n, 2017-07-08:
Will this work with the hx1000i? All I need is the wattage usage.

Thanks

jontis, 2017-07-08:
It's probably a fair chance that it does, but it needs to be told to use the RMI commands for it as it does not have info to identify it. Maybe the project could add an option like "--try-rmi" to try rmi commands on other corsair usb PSUs that are not yet identified as supported.

b0gd4n, 2017-07-08:
OK cool ... now .... how to I run this? :D

jontis, 2017-07-08:
It wont work, that option is just a suggestion, it is not implemented yet. However, I think it would be a 10 minute job for the authors of the project.

jontis, 2017-07-08:
but you can submit info for identifying you psu. 
use "lsusb" and copy the identifying string here.

audiohacked, 2017-07-08:
Submit the output for "lsusb -d 1b1c: -v"

audiohacked, 2017-07-08:
There's never going to be a "--try-rmi" option. Because lsusb will give us all the info we need to add support for the RM1000i.

squidgegmc, 2017-07-18:
I would like to help as i have the same PSU, here is the output:
`
Bus 001 Device 003: ID 1b1c:1c07 Corsair 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1b1c Corsair
  idProduct          0x1c07 
  bcdDevice            0.02
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      28
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
`

audiohacked, 2017-09-01:
Testing branch has the device added to the database. Please try the testing branch.

closed 2018-05-30 by audiohacked