hyperkineticnerd/OpenCorsairLink #33: H110i shows up with device id 0x00

hs123org, 2017-07-14:
I've confirmed the pump working on Windows and a Windows VM under Linux. So technically reading the data works.

lsusb shows the device:

Device: ID 1b1c:0c04 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c04 
  bcdDevice            2.00
  iManufacturer           1 Corsair Memory, Inc.
  iProduct                2 Integrated USB Bridge
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      37
          Report Descriptor: (length is 37)
            Item(Global): Usage Page, data= [ 0x00 0xff ] 65280
                            (null)
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x40 ] 64
            Item(Local ): Usage Minimum, data= [ 0x01 ] 1
                            (null)
            Item(Local ): Usage Maximum, data= [ 0x40 ] 64
                            (null)
            Item(Main  ): Input, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x40 ] 64
            Item(Local ): Usage Minimum, data= [ 0x01 ] 1
                            (null)
            Item(Local ): Usage Maximum, data= [ 0x40 ] 64
                            (null)
            Item(Main  ): Output, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0080
  (Bus Powered)

I've replaced the 0x41 device ID on device.c with 0x00 just to see if the USB communication itself would work and it's 'detected' as H110i as expected but no data can be read from USB.

Based on my order I got SKU CW-9060026-WW which is the plain H110i. 

debug output (at least part of it): 

usb device 4
corsair device 0
device_id 0x00
corsair device 1
device_id 0x00
corsair device 2
device_id 0x00
Dev=0, CorsairLink Device Found: H110i!
usb device 5

#  ./OpenCorsairLink.elf  --pump-mode 4 --device 0
Dev=0, CorsairLink Device Found: H110i!

Vendor: Corsair
Product: H110i
Firmware: 0.0.0.0
Temperature 0:  0.00
Temperature 1:  0.00
Temperature 2:  0.00
#

I'd be happy to snoop into the USB protocol under Windows if this helps identifying the device.

hs123org, 2017-07-14:
root@mars:~> usbhid-dump -d 1b1c:0c04
001:003:000:DESCRIPTOR         1500044002.528943
 06 00 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 19 01 29 40 81 02 09 01 75 08 95 40 19 01 29 40
 91 02 09 01 C0

CRCinAU, 2017-08-19:
I just got a h110i as well - and have the same problems. I see:

```
usb device 11
corsair device 0
device_id 0xFF
corsair device 1
device_id 0x83
corsair device 2
device_id 0x00
corsair device 3
device_id 0x00
corsair device 4
corsair device 5
corsair device 6
corsair device 7
corsair device 8
corsair device 9
corsair device 10
corsair device 11
```

I also get the same as you:
```
# usbhid-dump -d 1b1c:0c04
001:005:000:DESCRIPTOR         1503135548.281980
 06 00 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 19 01 29 40 81 02 09 01 75 08 95 40 19 01 29 40
 91 02 09 01 C0
```

mumblerit, 2017-09-01:
#  usbhid-dump -d 1b1c:0c04
001:002:000:DESCRIPTOR         1504263773.453998
 06 00 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 19 01 29 40 81 02 09 01 75 08 95 40 19 01 29 40
 91 02 09 01 C0

same issue

audiohacked, 2017-09-01:
I have no idea what I'm looking at when someone is posting usbhid-dumps....

CRCinAU, 2017-09-01:
What data would help in improving support?

mumblerit, 2017-09-01:
What information do you want? I was able to get my h110i to respond by changing the values in devices.c for the h110 to 
     .device_id = 0x00, /** H110i */

audiohacked, 2017-09-01:
I'm having trouble with my H100i, and it "appears" to work if the device_id is set to 0x00. This is really a problem with either the Hardware or Linux Kernel. The latest commits on both testing and master works in the Fedora 26 i686 machine I have. While my latest Archlinux install doesn't seem to fully work. As I have no way of testing every possible kernel or hardware variance, I'm going to close this issue until someone can provide info or a fix. As it stands, we already have all the information we could dump via lsusb.

closed 2018-05-30 by audiohacked