hyperkineticnerd/OpenCorsairLink #34: AX860i

chron0, 2017-07-14:
Thanks for giving us a linux alternative since Corsair won't. Unfortunately I couldn't get it to work with an AX860i over the PMBus/USB Adapter. My MB does have an I2C SMBus Powersupply port connected to BMC/IPMI - but since Corsair isn't even giving out the pinout of the I2C cable, I haven't even tried.

OpenCorsairLink just throws a segfault when running.

Here's the lsusb trace:

```
Bus 003 Device 004: ID 1b1c:1c00 Corsair Controller for Corsair Link
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1c00 Controller for Corsair Link
  bcdDevice            1.00
  iManufacturer           1 Silicon Labs
  iProduct                2 Corsair Link TM USB Dongle
  iSerial                 3 6525fguhngyh
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              2 Corsair Link TM USB Dongle
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

Am I doing it wrong or is the AX860i not supported? I'd really like to get temp/power metrics out of the PS to collect with https://github.com/firehol/netdata/wiki

audiohacked, 2017-07-14:
The AX860i isn't currently part of the "supported" device list. But it might be doable to add support if its the same protocol as the RM750i or other HXi PSUs

chron0, 2017-07-14:
If I can provide you with anything to help you, I'll be available over the weekend and can provide you with what ever debug data you require. Just give me a ping and what I should gather.

jdfergason, 2017-09-06:
I'd love to see this too.  I can help if needed but would need some guidance on where to start.  Here's the output of lsusb on my system:

```
Bus 001 Device 004: ID 1b1c:1c00 Corsair Controller for Corsair Link
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1c00 Controller for Corsair Link
  bcdDevice            1.00
  iManufacturer           1 Silicon Labs
  iProduct                2 Corsair Link TM USB Dongle
  iSerial                 3 0001ukhmnbvfjhnbjk
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 Corsair Link TM USB Dongle
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

Hagbard-Celin, 2017-11-16:
I also have this PSU and would like to be able to control it from linux. In case you have not noticed, I found some halfway-there code here: https://github.com/cloudkitsch/cpsumon (have not tried it myself, but have seen one report that it works for tuning fan to 100%)

AverageHumanoid, 2017-11-19:
+1

I'm using 2 AX860i per mining rig. I don't need to control anything but just want to get the AC input and DC output readings.

chron0, 2017-11-19:
same, no control needed, only power metrics are relevant to me.

chron0, 2017-11-19:
@Hagbard-Celin: Thnx for the link, I had a look into https://github.com/ka87/cpsumon - this is what we get out:

```
voyager /opt/cpsumon # ./cpsumoncli /dev/ttyUSB0
Corsair AXi Series PSU Monitor
(c) 2014 Andras Kovacs - andras@sth.sze.hu
-------------------------------------------

Dongle name: USB to SMB Bridge (Firmware by Ross Fosler)
Dongle version: 0.9
PSU type: AX860i
Fan speed: 0.00 RPM
Temperature: 39.75 °C
Voltage: 229.00 V
Current: 1.05 A
Input power: 229.87 W
Output power: 216.00 W
Efficiency: 93.97 %
PCIe 00 Rail:    12.02 V, 0.00 A, 0.00 W, OCP disabled (Limit: 40.00 A)
PCIe 01 Rail:    12.00 V, 0.00 A, 0.00 W, OCP enabled  (Limit: 39.94 A)
PCIe 02 Rail:    12.02 V, 4.88 A, 59.50 W, OCP enabled  (Limit: 39.94 A)
PCIe 03 Rail:    12.02 V, 0.00 A, 0.00 W, OCP enabled  (Limit: 40.00 A)
PCIe 04 Rail:    12.00 V, 0.00 A, 0.00 W, OCP enabled  (Limit: 39.94 A)
PCIe 05 Rail:    11.98 V, 8.81 A, 104.00 W, OCP enabled  (Limit: 40.00 A)
ATX Rail:        12.00 V, 0.00 A, 0.00 W, OCP disabled (Limit: 40.00 A)
Peripheral Rail: 12.02 V, 9.25 A, 109.50 W, OCP enabled  (Limit: 39.94 A)
5V Rail:         5.02 V, 3.00 A, 10.02 W
3.3V Rail:       3.30 V, 3.00 A, 5.70 W
```

okias, 2018-01-13:
Look good, thank you @chron0 

Would be nice if OpenCorsairLink would get this integrated.

Also I see duplicate issue for AX1200 #37

audiohacked, 2018-01-14:
@okias I'm sorry that I haven't been able to integrate the work of cpsumon, but I don't know how the protocol of the dongles work. If you could provide exactly how they work, I'd be more than welcome to add in support for them.

psblah, 2018-01-19:
for anyone else who wants to read data from a AX series PSU using https://github.com/ka87/cpsumon

if, like me, your usb device is missing you just need to bind the driver.

you need to bind the driver:

lsusb -t

you'll see output similar to this:

root@RIG1:~/cpsumon# lsusb -t
/: Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/: Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/: Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
/: Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
|__ Port 6: Dev 2, If 0, Class=Vendor Specific Class, Driver=rt2800usb, 480M
|__ Port 11: Dev 3, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M

only your Driver=cp210x will probably just say Driver=, 12M

you need to bind /dev/ttyUSB0 with the driver using the following command:

echo -n "1-11:1.0" > /sys/bus/usb/drivers/cp210x/bind

1-11:1.0 was my case because the device is on bus 1 port 11, substitute your bus / port number in place of my "1-11"

i also had to do echo -n "1-11:1.0" > /sys/bus/usb/drivers/cp210x/unbind

then

echo -n "1-11:1.0" > /sys/bus/usb/drivers/cp210x/bind

again before the device would answer.

then you device will appear in /dev/ttyUSB0

then you can use the command

~/cpsumon# ./cpsumoncli /dev/ttyUSB0
Corsair AXi Series PSU Monitor
(c) 2014 Andras Kovacs - andras@sth.sze.hu

Dongle name: USB to SMB Bridge (Firmware by Ross Fosler)
Dongle version: 0.9
PSU type: AX1200i
Fan mode: Auto
Fan speed: 0.00 RPM
Temperature: 43.75 °C
Voltage: 236.50 V
Current: 2.42 A
Input power: 566.39 W
Output power: 538.00 W
Efficiency: 94.99 %
PCIe 00 Rail:    11.92 V, 0.00 A, 0.00 W, OCP enabled  (Limit: 0.00 A)
PCIe 01 Rail:    11.97 V, 0.00 A, 0.00 W, OCP enabled  (Limit: 39.94 A)
PCIe 02 Rail:    11.97 V, 0.00 A, 0.00 W, OCP enabled  (Limit: 39.94 A)
PCIe 03 Rail:    11.91 V, 0.00 A, 0.00 W, OCP enabled  (Limit: 40.00 A)
PCIe 04 Rail:    11.97 V, 0.00 A, 0.00 W, OCP enabled  (Limit: 39.94 A)
PCIe 05 Rail:    11.97 V, 12.75 A, 151.00 W, OCP enabled  (Limit: 39.94 A)
PCIe 06 Rail:    11.97 V, 7.12 A, 86.00 W, OCP enabled  (Limit: 39.94 A)
PCIe 07 Rail:    11.97 V, 11.75 A, 139.00 W, OCP enabled  (Limit: 39.94 A)
ATX Rail:        11.97 V, 0.00 A, 0.00 W, OCP enabled  (Limit: 40.00 A)
Peripheral Rail: 11.95 V, 15.06 A, 179.00 W, OCP enabled  (Limit: 40.00 A)
5V Rail:         5.05 V, 0.00 A, 0.00 W
3.3V Rail:       3.28 V, 0.00 A, 0.00 W

closed 2018-05-30 by audiohacked

Patricol, 2018-07-08:
@audiohacked Is there a reason to close the issue? Leaving it open could grant more visibility; attracting the required assistance.

d3xt3r01, 2018-08-21:
I agree. Support for monitoring would be great. I got a 860i on one machine and was thinking of buying another one for the new machine.
If there's anything I can help with to get packets/debug/dump info and stuff .. let me know.

audiohacked, 2018-08-21:
At this moment, I'm not going to support the AXi PSUs for two major reasons. The Interface for the PSUs are using USB to SMB/Serial bridges, which is outside the scope of OpenCorsairLink. Two, nothing is known about the protocol for the PSUs, now I know there's software, but there's no description of the protocol used (who ever wrote cpusmon didn't comment or document whats going on). So until I have time or the money to buy an AXi powersupply and reverse engineer the two different physical interfaces this will never get implemented.

d3xt3r01, 2018-08-21:
Ok. If there's anything we can do to help you discover the protocol, let me know. I'm willing to test.