hyperkineticnerd/OpenCorsairLink #35: H110i GTX (non HID) support?

djurin, 2017-07-15:
Hi there, thank you for doing linux support for Corsair devices!

I am owner Corsair H110i GTX Extreme and I need to monitor and control it from Linux. I downloaded OpenCorsairLink and found that it does not recognize H110i GTX Extreme. Quickly looking into the code, I found that my cooler does not fall under any of your device definitions. I added a new entry to corsair_device_info corsairlink_devices structure based on the data from lsusb:

```
{
   	 .vendor_id = 0x1b1c,
   	 .product_id = 0x0c07,
   	 .device_id = 0xFF,
   	 .name = "H110i GTX", /** H110i GTX */
   	 .read_endpoint = 0x02|LIBUSB_ENDPOINT_IN,
   	 .write_endpoint = 0x02|LIBUSB_ENDPOINT_OUT,
   	 .driver = &corsairlink_driver_asetek,
   	 .led_control_count = 1,
   	 .fan_control_count = 1,
   	 .pump_index = 0,
},
```
So, with this it works fine and I can see some info and some temperature 
```
Dev=0, CorsairLink Device Found: H110i GTX!
Vendor: Corsair
Product: H110i GTX
Firmware: 2.4.0.0
Temperature 0: 30.30
```
and even to change the pump mode: 
```
sudo ./OpenCorsairLink.elf --device 0 --pump-mode 5
```
Also I see that I can change FanCurves and LED color which is nice,  but is there a way to retrieve the rest of information that provided by CorsairLink like pump and fan rpm-speeds ?

djurin, 2017-07-23:
I just noticed that after recent updates my Corsair H110i GTX cooler is now recognised by OpenCorsairLink ! Moreover the testing brunch now is showing fan and pump speeds !!!! which is totally awesome! Thanks a lot! 

But two thing are unclear for me, first is the temperature reading that I obtain with OpenCorsairLink which is giving me always numbers like 20-32, while my CPU temps varying from 20 to 80 C. What is this this temperature? is it radiator temperature? Or liquid temperature in the pipes?

Second, how can I setup a custom pump-speeds curve like it is done with fan-speeds? It looks in the code that it is possible. It would totally nice if we will be able to do like this :

`sudo /OpenCorsairLink.elf --device 0 --pump-speeds 25,75,100,100,100,100 --pump-temps 25,27,40,50,60,70`

Because right now, when I set pump-mode to 5 it goes to its max speed 2850 or so and stay there independently of the pump or cpu temperatures, if I set it to 3 or 4 it goes to 1900 or so and stay fixed to it also independently of the pump or cpu temperatures.

SmileCafe, 2017-12-10:
I'm a new user in this site. Where do I find the manual for this OpenCorsairLink?? Thanks for reading.

closed 2018-05-30 by audiohacked