hyperkineticnerd/OpenCorsairLink #37: AX1200i Support Request

djurin, 2017-07-17:
Once you started with Corsair Link USB Dongle for AX860i so maybe you can at the same time add AX1200i ? :) Here's corresponding lsusb output:

```
Bus 003 Device 002: ID 1b1c:1c00 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1c00 
  bcdDevice            1.00
  iManufacturer           1 Silicon Labs
  iProduct                2 Corsair Link TM USB Dongle
  iSerial                 3 R57U0633
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 Corsair Link TM USB Dongle
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```
If anything is needed from my side an additional info or testing just let me know. Thanks!

marcoleewow, 2017-10-20:
does the current version work with AX1200i? @djurin

djurin, 2017-10-21:
Nope.
```
$ sudo ./OpenCorsairLink.elf --device 0
[sudo] password for yurin: 
Dev=0, CorsairLink Device Found: USB Dongle!
Dev=1, CorsairLink Device Found: H110i GTX!

libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=22
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=22
Vendor: 
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=22
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=22
Product: 
Firmware: NA
Segmentation fault (core dumped)
```

audiohacked, 2017-10-21:
That's because the driver hasn't been written yet. I haven't had much time to do the work for it.

cjustin88, 2017-11-16:
I have an AX760i. When you have the time and are interested in working on supporting the AX series, I am happy to provide whatever hardware information I can.

closed 2018-05-30 by audiohacked