hyperkineticnerd/OpenCorsairLink #38: Running OpenCorsairLink.elf with only `--device` resets LED color for H115i

nhinds, 2017-07-24:
I have a H115i (`1b1c:0c0a`) and ran `sudo ./OpenCorsairLink.elf --device 0` to see what this app knew about my cooler.

The output seems correct:

```
$ sudo ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H115i!

Vendor: Corsair
Product: H115i
Firmware: 2.8.0.0
Temperature 0: 36.10
```

but running this command modified my cooler's LED from the color I'd set in Windows to a dark grey. I would normally expect that just running the command without specifying `--led` would not modify any LED settings.

I have not confirmed whether it also modified pump/fan settings.

nhinds, 2017-07-24:
The LED and pump/fan settings were all back to normal after powering the system off and back on, so whatever changes were made appear to only be temporary.

audiohacked, 2017-07-24:
As your H115i is based on Asetek, the current way of getting info is to send a "change led color" command, the response of the command is all the stats you normally get from CorsairLink.

denysvitali, 2017-08-13:
I'm pretty sure the problem is [here](https://github.com/audiohacked/OpenCorsairLink/blob/master/main.c#L150). Can you please try to comment it out and see if it happens again?

audiohacked, 2017-08-13:
@denysvitali Your fix isn't going to help for the H115i as the Asetek protocol doesn't remotely work similarly to the HID protocol. The HID protocol has a command to retrieve stats from the cooler but the Asetek protocol ties the stats command with the change LED command.

denysvitali, 2017-08-13:
I see.  
  
There are two different issues then:  
- OP states that when he selects a device, his H115i (Asetek) changes from a color he chosen before, to a dark grey color.  
- In my case, when I run the same command: `/OpenCorsairLink.elf --device 0` with my H80i, my LED configuration gets reset, due to [this line](https://github.com/audiohacked/OpenCorsairLink/blob/master/main.c#L150), as previously mentioned.  
  
Now, I understand your point, and I've seen how the Asetek protocol is implemented in OpenCorsairLink (specifically [the comand set requires a LED color](https://github.com/audiohacked/OpenCorsairLink/blob/master/protocol/asetek4/core.c#L100-L118)) but if the comments are right, the LED should turn green or red, not dark grey.  
  
Unfortunately I don't have an H115i to test it or analyze the protocol via an USB packet sniffing - therefore my knowledge on this particular issue is limited to what I know about OpenCorsairLink, the HID protocol, and my personal experience.

closed 2018-05-30 by audiohacked