hyperkineticnerd/OpenCorsairLink #43: Support RM1000i

lu-zero, 2017-08-05:
Here the lsusb

```
lsusb -d 1b1c:1c0d -v

Bus 001 Device 002: ID 1b1c:1c0d Corsair
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x1b1c Corsair
  idProduct          0x1c0d
  bcdDevice            0.02
  iManufacturer           1
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      28
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

lu-zero, 2017-08-05:
Just adding the device descriptor to the list does not seem to produce a valid output:
``` diff
diff --git a/device.c b/device.c
index 60d0df9..b89cdbc 100644
--- a/device.c
+++ b/device.c
@@ -165,6 +165,18 @@ struct corsair_device_info corsairlink_devices[] = {
                .fan_control_count = 1,
                .pump_index = 0,
        },
+       {
+               .vendor_id = 0x1b1c,
+               .product_id = 0x1c0d,
+               .device_id = 0xFF,
+               .name = "RM1000i",
+               .read_endpoint = 0x01|LIBUSB_ENDPOINT_IN,
+               .write_endpoint = 0x01|LIBUSB_ENDPOINT_OUT,
+               .driver = &corsairlink_driver_rmi,
+               .led_control_count = 0,
+               .fan_control_count = 1,
+               .pump_index = 0,
+       },
 };

 size_t corsairlink_device_count = sizeof(corsairlink_devices)/sizeof(corsairlink_devices[0]);
```
```
DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor:
Product:
Firmware: NA
DEBUG: string done
03 FE 00 00 00 00
Temperature 0:  0.00 C
03 FE 00 00 00 00
Temperature 1:  0.00 C
Powered: 0 (0d.  0h)
Uptime: 0 (0d.  0h)
DEBUG: time done
03 FE 00 00 00 00
Supply Voltage:  0.00
03 FE 00 00 00 00
Total Watts:  0.00
DEBUG: supply done
Output 12v:
03 FE 00 00 00 00
	Voltage  0.00
03 FE 00 00 00 00
	Amps  0.00
03 FE 00 00 00 00
	Watts  0.00
Output 5v:
03 FE 00 00 00 00
	Voltage  0.00
03 FE 00 00 00 00
	Amps  0.00
03 FE 00 00 00 00
	Watts  0.00
Output 3.3v:
03 FE 00 00 00 00
	Voltage  0.00
03 FE 00 00 00 00
	Amps  0.00
03 FE 00 00 00 00
	Watts  0.00
```

audiohacked, 2017-08-23:
I've noticed that a few people are getting "03 FE 00 00 00 00" instead of the proper output:
```
usb device 0
corsair device 0
corsair device 1
corsair device 2
corsair device 3
corsair device 4
corsair device 5
corsair device 6
corsair device 7
corsair device 8
corsair device 9
corsair device 10
corsair device 11
usb device 1
corsair device 0
corsair device 1
corsair device 2
corsair device 3
corsair device 4
corsair device 5
corsair device 6
corsair device 7
corsair device 8
corsair device 9
device_id 0xFF
Dev=0, CorsairLink Device Found: RM750i!
usb device 2
corsair device 0
corsair device 1
corsair device 2
corsair device 3
corsair device 4
corsair device 5
corsair device 6
device_id 0xFF
Dev=1, CorsairLink Device Found: H100i V2!
usb device 3
corsair device 0
corsair device 1
corsair device 2
corsair device 3
corsair device 4
corsair device 5
corsair device 6
corsair device 7
corsair device 8
corsair device 9
corsair device 10
corsair device 11

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: CORSAIR
Product: RM750i
Firmware: NA
DEBUG: string done
03 8D C9 F0 00 00
Temperature 0: 50.50 C
03 8E A0 F0 00 00
Temperature 1: 40.00 C
Powered: 17381115 (201d.  4h)
Uptime: 1602315 (18d.  13h)
DEBUG: time done
03 88 E6 F8 00 00
Supply Voltage: 115.00
03 EE 3F 08 00 00
Total Watts: 128.00
DEBUG: supply done
Output 12v:
03 8B 03 D3 00 00
	Voltage 12.06
03 8C 1E F0 00 00
	Amps  7.50
03 96 2B 08 00 00
	Watts 88.00
Output 5v:
03 8B 3F D1 00 00
	Voltage  5.00
03 8C 3B E0 00 00
	Amps  3.75
03 96 24 F8 00 00
	Watts 18.00
Output 3.3v:
03 8B D3 D0 00 00
	Voltage  3.31
03 8C 1C E0 00 00
	Amps  1.75
03 96 0B F8 00 00
	Watts  6.00
```
if you ever figure out why its not working, I would love to add support for it and other PSUs.

audiohacked, 2017-09-01:
I figured out why the PSU was returning "03 FE 00 00 00". There should be a fix on master now!

closed 2018-05-30 by audiohacked

viggy96, 2019-08-15:
OpenCorsairLink no longer provides output for the RM1000i.

viggy96, 2019-08-22:
OpenCorsairLink does not see the RM1000i.