hyperkineticnerd/OpenCorsairLink #45: HX750i PSU support

mdlayher, 2017-08-19:
Hello!  Just got a new Linux machine with an HX750i and trying to give this library a try before attempting to write on in Go.

It appears that @awoland added HX850i, so I've tried to copy that device's configuration and modified the product ID to fit the HX750i (0x1c05).  OCL is able to pick up the device, but it appears that all queries to the device are returning `03 FE 00 00 00 00`. I noticed a similar behavior when attempting to talk with the device from Go.

Is some kind of initialization handshake necessary to communicate with the device?

Here's what I've added so far:

```patch
diff --git a/device.c b/device.c
index 60d0df9..4dfd2c2 100644
--- a/device.c
+++ b/device.c
@@ -164,7 +164,19 @@ struct corsair_device_info corsairlink_devices[] = {
                .led_control_count = 0,
                .fan_control_count = 1,
                .pump_index = 0,
-       },      
+       },
+       {
+               .vendor_id = 0x1b1c,
+               .product_id = 0x1c05,
+               .device_id = 0xFF,
+               .name = "HX750i",
+               .read_endpoint = 0x01|LIBUSB_ENDPOINT_IN,
+               .write_endpoint = 0x01|LIBUSB_ENDPOINT_OUT,
+               .driver = &corsairlink_driver_rmi,
+               .led_control_count = 0,
+               .fan_control_count = 1,
+               .pump_index = 0,
+       },
 };
 
 size_t corsairlink_device_count = sizeof(corsairlink_devices)/sizeof(corsairlink_devices[0]);

```

And here are a couple of gists with `lsusb` and OCL output:

`sudo lsusb -d 1b1c: -v > lsusb.txt`
https://gist.github.com/mdlayher/9357a20cb2f1252788b87ae5c257f186

`sudo ./OpenCorsairLink.elf --debug --device 0 > ocl.txt`
https://gist.github.com/mdlayher/ebc53483b285459015f7fec44c454157

If anyone reading is able to provide some insight, I'd greatly appreciate it. Thanks for your time.

kgara, 2017-08-29:
Confirm that with RM750i - all zero in output, yet it is not :(

audiohacked, 2017-08-29:
@kgara What?

I just ran OpenCorsairLink with my RM750i and it works correctly!

kgara, 2017-08-29:
From root
```# ./OpenCorsairLink.elf --debug --device 0 > /var/log/opencorsairlink.01.txt```
[opencorsairlink.01.txt](https://github.com/audiohacked/OpenCorsairLink/files/1260127/opencorsairlink.01.txt)

mdlayher, 2017-08-29:
That looks like the same thing I'm seeing with the HX750i.  Very interesting. @audiohacked, I am curious if there is some initialization that needs to happen. It feels weird that the two different devices return the same output that doesn't produce valid results.

Maybe the Corsair Link software in Windows modifies the configuration in some way? I've never run windows on this machine, but I could potentially pass the device through to a Windows VM.

kgara, 2017-08-29:
> Maybe the Corsair Link software in Windows modifies the configuration in some way?

Sound strange, yet not impossible.

> I've never run windows on this machine

I did but without Corsair Link being installed, may try it out in a few hours.

mdlayher, 2017-08-29:
Please do report back if you hear anything. I'd love to find an answer.

jontis, 2017-08-29:
Might be the same I'm seeing for RM650i. Device is found and identified correctly but returns only 0 when queried.

audiohacked, 2017-08-30:
Is everyone running as root? or using Sudo?

mdlayher, 2017-08-30:
```
matt $ ./OpenCorsairLink.elf
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/005: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.

matt $ sudo ./OpenCorsairLink.elf
Dev=0, CorsairLink Device Found: HX750i!

matt $ sudo su
root # ./OpenCorsairLink.elf 
Dev=0, CorsairLink Device Found: HX750i!

```

Doesn't work as non-root, and I get the same output as before.

```
$ sudo ./OpenCorsairLink.elf --device 0 --debug
# some lines trimmed
usb device 8
corsair device 0
corsair device 1
corsair device 2
corsair device 3
corsair device 4
corsair device 5
corsair device 6
corsair device 7
corsair device 8
corsair device 9
device_id 0xFF
Dev=0, CorsairLink Device Found: HX750i!

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: 
Product: 
Firmware: NA
DEBUG: string done
03 FE 00 00 00 00
Temperature 0:  0.00 C
03 FE 00 00 00 00
Temperature 1:  0.00 C
Powered: 0 (0d.  0h)
Uptime: 0 (0d.  0h)
DEBUG: time done
03 FE 00 00 00 00
Supply Voltage:  0.00
03 FE 00 00 00 00
Total Watts:  0.00
DEBUG: supply done
Output 12v:
03 FE 00 00 00 00
        Voltage  0.00
03 FE 00 00 00 00
        Amps  0.00
03 FE 00 00 00 00
        Watts  0.00
Output 5v:
03 FE 00 00 00 00
        Voltage  0.00
03 FE 00 00 00 00
        Amps  0.00
03 FE 00 00 00 00
        Watts  0.00
Output 3.3v:
03 FE 00 00 00 00
        Voltage  0.00
03 FE 00 00 00 00
        Amps  0.00
03 FE 00 00 00 00
        Watts  0.00
```

No luck so far, @audiohacked.

audiohacked, 2017-08-30:
It's definitely NOT a different protocol, the Corsair Link 4 software only has 4 device protocols: Flextronics PSUs, Asetek Coolers, old HID devices, and the Commander Pro devices. It could be firmware.

mdlayher, 2017-08-30:
I know there was a thread on the Corsair forums about the Link protocol. I'll reach out there tomorrow and see what I can find out.

audiohacked, 2017-08-30:
That thread really only talks about the HID protocol and none of the others.

kgara, 2017-08-30:
Minor update from my side.
1. Same behavior on my other  same RM750i.
2. It !was WORKING! I (actually person I asked, those PSU stays thousands km from me) rebooted into windows installed and started Corsair Link, rebooted (without pc power down) and it started working in Linux, both PSU.
```
...
Uptime: 6240 (0d.  1h)
Supply Voltage: 230.00
Total Watts: 208.00
...
```
Then I made shutdown. We removed the power. Restarted and got zero again. Tried one more time with Windows, but it did not work (in Linux zeros again). Moreover person reports Corsair Link to crash and disappear from tray.
Strange. But obviously some initialization logic made in Windows.
Will try to experiment some more.

audiohacked, 2017-09-01:
I figured out the problem with all the HXi and RMi PSU "03 FE 00...." responses. Apparently the PSUs expect at least one "03 FE" command before any query of stats. Patch/commit in comming to master.

mdlayher, 2017-09-02:
@audiohacked you are my hero!! Thank you so much!!

```
matt@servnerr-3:~/tmp/OpenCorsairLink⟫ sudo ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: HX750i!

Vendor: CORSAIR
Product: HX750i
Firmware: NA
Temperature 0: 45.00 C
Temperature 1: 38.50 C
Powered: 1156257 (13d.  9h)
Uptime: 1062657 (12d.  7h)
Supply Voltage: 115.00
Total Watts: 60.00
Output 12v:
        Voltage 12.09
        Amps  3.50
        Watts 40.00
Output 5v:
        Voltage  4.97
        Amps  3.25
        Watts 16.00
Output 3.3v:
        Voltage  3.28
        Amps  2.38
        Watts  8.00
```

kgara, 2017-09-03:
Works for me also. Thanks man! Nice and useful tool, keep going!

kgara, 2017-09-03:
Of-topic: Is there an out of box metric for statistical power consumption. E.g. 1.5 kWatt Hours for last month or for current up-time. Or should I call OCL from cron, parse and count myself? Thanks in advance.

audiohacked, 2017-09-06:
No metric saving, you'll have to figure out a script to parse stats.

closed 2018-05-30 by audiohacked