hyperkineticnerd/OpenCorsairLink #47: Source of uint16 to double formula

mdlayher, 2017-09-02:
Hello!  I was spelunking through the source code today when I noticed the function `convert_bytes_double`.

https://github.com/audiohacked/OpenCorsairLink/blob/master/protocol/rmi/core.c#L35

Can you explain how this formula was obtained?  I've ported it to Go and it appears to work fine in my programs as well, though I haven't the faintest idea why.

Thanks for your time.

audiohacked, 2017-09-02:
It's the standard algorithm to convert two bytes into a double.

mdlayher, 2017-09-02:
Can you provide any links?  Did a fair amount of Googling last night and couldn't find anything that resembled it.

audiohacked, 2017-09-02:
https://en.wikipedia.org/wiki/Half-precision_floating-point_format

closed 2018-05-30 by audiohacked