hyperkineticnerd/OpenCorsairLink #48: RMi driver fan speed retrieval

mdlayher, 2017-09-02:
https://github.com/audiohacked/OpenCorsairLink/blob/master/protocol/rmi/fan.c#L22

It appears that fan speed retrieval functionality is currently disabled for the RMi driver.  Is there a reason why?

audiohacked, 2017-09-02:
The majority of the time the Fan RPM of the RMi will be zero. I haven't used the function so I didn't notice I didn't finish the function.

mdlayher, 2017-09-02:
Seems like it works fine when enabled.  Should this be reopened to track that it's not enabled?

audiohacked, 2017-09-02:
It's disabled because I know people are going to freak out if I enabled it and their PSU's fan speed is reading zero (which even while gaming it won't turn on). The RMi/HXi/AXi are currently designed to not run the Fan unless the PSU usage is over 60? percent; not even during my gaming sessions have I had my RM750i's fan turn on.

closed 2018-05-30 by audiohacked