hyperkineticnerd/OpenCorsairLink #50: h115i fan issue

wmilas, 2017-09-23:
I can set the pump no problem. Toggling between 3, 4, 5 sets the pump speed. The problem is setting the fan speeds and temps. I look at the options header and I see can both are six values.

./OpenCorsairLink.elf --device 0 --fan-speeds 40,50,60,70,80,100
./OpenCorsairLink.elf --device 0 --fan-temps 20,25,29,33,37,40

Both those seem to do absolutely nothing. No error, no change in fan speed.

Dev=0, CorsairLink Device Found: H115i!

Vendor: Corsair
Product: H115i
Firmware: 2.8.0.0
Temperature 0: 34.20
Fan Speed 0: 2460
Pump Speed: 2880

No matter if I set fan-speeds to something like 40,40,40,40,40,40 It doesn't matter. Fans are pegged and running at 100%.

In windows the speeds are tied to profiles and you have to set the profile. Do I somehow have to set the profile to custom to get the temps and speeds to be recognized?

Thanks!

audiohacked, 2017-09-28:
You're not setting both 'fan-speeds' and 'fan-temps' together.

wmilas, 2017-10-01:
Bingo. thanks.

closed 2018-05-30 by audiohacked