hyperkineticnerd/OpenCorsairLink #54: Can't change speeds and temps after first setting on H100i v2

eera5607, 2017-10-30:
Hi!
When I specify a set of temps and speeds the first time it apparently works fine. I can notice the change of fans speeds after setting a "performance mode"
`./OpenCorsairLink.elf --device 0 --fan-temps 20,25,29,33,37,40 --fan-speeds 1380,1680,1920,2220,2400,2820`
the thing is that now I can not go back to the quiet mode, I mean, the fans speed up at maximum speed and can't go back to the predetermined state. It seems to ignore the commands after that. 

The first time I change the "profile", what ever values I use it kind of get it right, the subsequent commands are "ignored" by the program. I have to restart the PC, go to Windows and change the profile on Corsair Link and it starts working fine again (at least the first command I use).

audiohacked, 2017-11-01:
You're doing fan speeds all wrong. The H100i v2 takes percentages, not RPMs.

eera5607, 2018-05-17:
Old commands that I used to use don't seem to be working, for example:

sudo ./OpenCorsairLink.elf --device 0 --fan-temps 20,25,29,33,37,40 --fan-speeds 40,50,60,70,80,100

Default fan modes also not working on H100i v2.

eera5607, 2018-05-30:
Vendor: Corsair
Product: H100i V2
Firmware: 2.8.0.0

For some reason, the pump mode affects the previously selected fan mode. If I select the _Performance_ fan mode, and change the pump mode to _Silent_ the fans go down to 900 RPM. 

Sometimes the fan modes work sometimes don't.

closed 2018-08-01 by audiohacked