hyperkineticnerd/OpenCorsairLink #56: H80i (HID) returns 0 for temp, fan and speed - due to 64 vs 32 byte messages?

n1c0la5, 2017-11-18:
It shows:
Vendor: Corsair
Product: H80i
Firmware: 0.59.0.0
Temperature 0:  0.00
Temperature 1:  0.00
Temperature 2:  0.00
Fan Speed 0: 0
Fan Speed 1: 0
Fan Speed 2: 0
Fan Speed 3: 0
Pump Speed: 0

Looking at the code in protocol/hid/*.c some message lengths are 32 instead of 64. Based on http://forum.corsair.com/v3/showthread.php?t=120092&page=4 the messages are all to 64 bytes, not sure if that is intentional/works for other models...

So in protocol/hid/*.c made the following changes : 
	uint8_t response[64];
	uint8_t commands[64];
and the same for the length parameter in the dev->driver->read() calls.

Then it gives me:

Vendor: Corsair
Product: H80i
Firmware: 0.59.0.0
Temperature 0:  0.00
Temperature 1:  0.00
Temperature 2:  0.00
Fan Speed 0: 33030
Fan Speed 1: 33542
Fan Speed 2: 34054
Fan Speed 3: 34566
Pump Speed: 59

audiohacked, 2017-11-25:
I can already tell that your fix is incorrect. One, no fan will have a fan RPM speed of 33,000. Two the transfer size of 32 or 64 is irrelevant since all commands are less than 32.

audiohacked, 2017-11-25:
Disregard my previous comment. Double checking my previous code shows the problems.

closed 2018-05-30 by audiohacked