hyperkineticnerd/OpenCorsairLink #57: H80i shows 0's for temperature as there is no temperature to celsius conversion for hid

n1c0la5, 2017-11-18:
In main.c 141 it only does temperature to celsius for asetek.
```
if (dev->driver == &corsairlink_driver_asetek) {
			uint8_t v1 = (temperature>>8);
			uint8_t v2 = (temperature&0xFF);
			msg_debug("DEBUG: %02X %02X\n", v1, v2);
			celsius = (double)v1 + ((double)v2/10);
		}
```

After commenting out lines 141 and 146 I get:
```
Vendor: Corsair
Product: H80i
Firmware: 0.59.0.0
Temperature 0: 81.90
Temperature 1: 64.60
Temperature 2:  0.00
Fan Speed 0: 33030
Fan Speed 1: 33542
Fan Speed 2: 34054
Fan Speed 3: 34566
Pump Speed: 59
```

Whereas before it was:
```
Vendor: Corsair
Product: H80i
Firmware: 0.59.0.0
Temperature 0:  0.00
Temperature 1:  0.00
Temperature 2:  0.00
Fan Speed 0: 33030
Fan Speed 1: 33542
Fan Speed 2: 34054
Fan Speed 3: 34566
Pump Speed: 59
```

audiohacked, 2017-11-25:
Temperatures are already in Celsius coming from the H100i/H80i.

audiohacked, 2017-11-25:
To convert to a float the need for dividing 256 is needed.

closed 2018-05-30 by audiohacked