hyperkineticnerd/OpenCorsairLink #59: Corsair H115i

source83, 2017-11-19:
It works, and reads out data, but every time the executable is run and it outputs the data, I can hear the fan and/or the pump rev up then fall back to normal very fast.

I was originally planning on using this in consort with conky (hw monitoring +++) where the executable would be run 'n' times a second or every 'n' seconds.

This went well until I tried to slowly have it update faster where the fan and pump suddenly went full blast until I rebooted since I could not access the usb device anymore.

I was planning on trying to filter out what was needed to build a smaller more directed version of your code for my H115i so it didn't need to scan.

Any ideas what this all could be related to ?

audiohacked, 2017-11-25:
The software isn't designed for atomic access to hardware. The software is also not designed for temperature monitoring, more of led and fan tweaking.

source83, 2017-11-26:
Okay, is it possible to reuse some of your code to build something that could handle monitoring ?
Or is the code to "invasive" on the functions of the pump to be reused in this manner ?

mmccool, 2018-02-01:
I have seen hangs as well.   I was running the program periodically using "watch" but then was starting to get errors access the device (invalid bulk transfer).   A "ps -e" revealed a bunch of  OpenCorsairLink zombies: processes still in memory, but that I was unable to kill.  I had to reboot to get rid of them.

I have both a H112i and a HX1200i though.   I saw more problems when I tried accessing these two devices asynchronously in separate processes; accessing them sequentially in one script seemed to work better.   But I still had the above problem with processes hanging and failed bulk transfers.

closed 2018-05-30 by audiohacked