hyperkineticnerd/OpenCorsairLink #68: H100i v2 Temperature not detecting correctly

AbysmalBiscuit, 2018-01-05:
Hi,

I have an H100i v2 cooler and OpenCorsairLink (OCL) seems to fail to correctly  read its temperature.
For instance when running a sysbench test, lm_sensors detects temperature ranges of around ~80C and my case fans ramp up (since I configure their fan curves in the BIOS), however OCL shows temperatures that start from ~30C and very slowly increase the longer I run the benchmark. After stopping the benchmark the reported temperature takes some time to decrease. This is making me think that perhaps OCL is detecting the liquid temperature and not the cpu temperature.

With regard to all the other functionality OCL seems to work fine, I can change the led color, warning led color and temperature, fan curves and pump speed.

I made an attempt to investigate the cause and thus I had modified the `corsairlink_asetek_temperature()` function in `protocol/asetek4/temperature.c` to print out the entire `response` array, but none of the values seemed to match those reported by lm_sensors (this is unfortunately my limit in understanding C so I don't know what to do beyond this).

Screenshot of benchmark and temperature readings:
[image]
Top left: OCL readings, I waited several seconds before re-running the command.
Bottom left: sysbench benchmark.
Right: lm_sensors readout.

DroWnThePoor, 2018-01-08:
I'm about to install this same model, Wondering if this cooler will work without OpenCorsairLink? Is this driver simply to supply missing features?

audiohacked, 2018-01-09:
The reason lm_sensors reads differently than OCL is because lm_sensors reads off the CPU die where as OCL reads the water flow temperature.

audiohacked, 2018-01-09:
Also, the temperature reading off the CPU will be much more volatile and in-accurate to real-world temperatures.

audiohacked, 2018-01-09:
One last thing, OCL isn't needed for the water coolers, the default settings are more than enough to handle stock CPU speeds and cooling needs. OCL and Corsair's software are only useful for changing LED colors and fan speeds.

AbysmalBiscuit, 2018-01-10:
I see, thank you for taking the time to explain.
My CPU is overclocked hence why I wanted to make sure I understood whether the fan curves should scale with CPU or liquid temperatures.

I have one last follow up question, does this mean that the Windows Corsair Link also uses the liquid temperatures for its fan curves/warning led settings?

audiohacked, 2018-01-10:
Yes, Windows Corsair Link also uses the liquid temperatures. If you are seeing a large discrepancy between the CPU temperature and the Water temperature, you might not have good thermal bonding between the Water block and the CPU. Double check your mounting.

closed 2018-08-01 by audiohacked