hyperkineticnerd/OpenCorsairLink #70: Support for Corsair Commander Pro (I figured out the protocol)

pvh1987, 2018-01-15:
Hello :-)

I am so glad I found OpenCorsairLink. I recently decided to build a new workstation. I purchased the H115i as well as the HX850i - I guess that was a lucky choice because, as I later found out, OpenCorsairLink supports both devices already. However, I also got a Corsair Commander Pro that works with Corsair LINK as well. I use this to control various fans in my PC case.

Instead of being disappointed that there is currently no support for the Commander Pro in OpenCorsairLink, I decided to figure out the protocol myself before I post anything here on GitHub. I succeeded, well almost - there is still a lot of RGB LED related stuff I have yet to figure out - but the most important things like fan control and temperature measurement are clear by now. I have included a very tiny test program to show that it works.

I hope that the protocol details below are enough to implement support for this device in OpenCorsairLink. Then soon, I can have my workstation just as silent in Linux as in M$ Windows. Thanks in advance :-)

Oh, by the way - what is the future goal of OpenCorsairLink? I get that at the moment, it is just a small tool to set up fans and the like, with no real monitoring support. I know that the official LINK software is able to monitor and inform devices, such as the Corsair Commander Pro, about temperature. That way, the temperature of a H115i or HX850i can control the fans connected to the Commander Pro. I guess a possible solution is to create a simple daemon that can run at boot time, constantly monitoring temperatures (at a set interval) and also have a small API that can be used to connect to GUI apps or panel applets, like the "MATE Sensors Applet" and similar. I think this is better than to connect/disconnect to the USB devices all the time directly... and what happens if more users are logged in at the same time, both monitoring/controlling Corsair LINK devices? Well... just some of my thoughts :-)

Please let me know if I can be helpful in more ways. I'm an electronics engineer and programmer, but I have little to no experience in Unix/Linux system programming. I do, however, have quite a bit experience with cross-platform desktop GUI programming (mainly C++ with wxWidgets) - so if a GUI for OpenCorsairLink is needed, I might be willing to contribute if I can find the time.

Anyway, still thanks - and below are my findings regarding the protocol and workings of the Corsair Commander Pro :-)

---

Introduction
=========

The Corsair Commander Pro is a device that can connect to
- 6 fans (3- or 4-pin)
- 4 temperature sensors (included)
- 2 RGB LED channels
- 2 internal USB devices such as Corsair PSUs and CPU water coolers (acts like a hub)

The device can also measure the rail voltages from the PSU (supplied by a SATA power connector).
Each of the six fans can be monitored (RPM measurement).

The device connects to an internal USB connector on the motherboard.

Link: http://www.corsair.com/en-us/corsair-commander-pro

I have sniffed the protocol using a VirtualBox with Windows 7 and the so called "Free USB Analyzer" from HDD Software. BTW, it is actually not free software.
Link: https://freeusbanalyzer.com/

The protocol is described below. Additional notes are found below in the NOTES section.

IMPORTANT: The LED functionality of this device is complicated. I have not been able to figure out what is really going on with most of the LED commands. Some modes seems to work in a completely different way than others - some modes only transmits commands when applying settings, other modes "spam" the device with commands all the time. So much more research is needed here. Since I do not intend to use the LED channels myself, I doubt I can find the time to do this. My experiments so far with USB packet sniffing have already been very time consuming. I have tried to document some of it below, especially in the "Additional notes for LED channel operation".

Device Information (lsusb)
==========================

Bus 001 Device 005: ID 1b1c:0c10 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1b1c Corsair
  idProduct          0x0c10 
  bcdDevice            0.04
  iManufacturer           1 Corsair
  iProduct                6 Commander PRO
  iSerial               128 0905031174181010
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
[... 401 lines cut ...]
    res = hid_write(handle, outbuf, 63);
    res = hid_read(handle, inbuf, 16);

    // Print result
    for (i = 0; i < 16; ++i)
        printf("%x ", inbuf[i]);
    printf("\n");

    outbuf[0] = 0x23;   // Set fan to fixed %
    outbuf[1] = 0x00;   // Select fan number 1
    outbuf[2] = 0x64;   // Set speed to 100%
    res = hid_write(handle, outbuf, 63);
    res = hid_read(handle, inbuf, 16);

    sleep(5);

    outbuf[0] = 0x23;   // Set fan to fixed %
    outbuf[1] = 0x00;   // Select fan number 1
    outbuf[2] = 0x1E;   // Set speed to 30%
    res = hid_write(handle, outbuf, 63);
    res = hid_read(handle, inbuf, 16);

    return 0;
}
```

audiohacked, 2018-01-16:
Nice! I was planning on reverse engineering the Pro protocol, but been really busy.

PostalBlab, 2018-04-10:
Here are all available pro commands, hope it helps:
ReadStatus = 0x01;
ReadFirmwareVersion = 0x02;
ReadDeviceId = 0x03;
WriteDeviceId = 0x04;
StartFirmwareUpdate = 0x05;
ReadBootloaderVersion = 0x06;
WriteTestFlag = 0x07;

ReadTemperatureMask = 0x10;
ReadTemperatureValue = 0x11;
ReadVoltageValue = 0x12;

ReadFanMask = 0x20;
ReadFanSpeed = 0x21;
ReadFanPower = 0x22;
WriteFanPower = 0x23;
WriteFanSpeed = 0x24;
WriteFanCurve = 0x25;
WriteFanExternalTemp = 0x26;
WriteFanForceThreePinMode = 0x27;
WriteFanDetectionType = 0x28;
ReadFanDetectionType = 0x29;

ReadLedStripMask = 0x30;
WriteLedRgbValue = 0x31;
WriteLedColorValues = 0x32;
WriteLedTrigger = 0x33;
WriteLedClear = 0x34;
WriteLedGroupSet = 0x35;
WriteLedExternalTemp = 0x36;
WriteLedGroupsClear = 0x37;
WriteLedMode = 0x38;
WriteLedBrightness = 0x39;
WriteLedCount = 0x3a;
WriteLedPortType = 0x3b;

audiohacked, 2018-04-10:
The only barrier I see, is changing the Driver API to accommodate the expanded RGB system in CL.

pvh1987, 2018-04-10:
I have worked on reverse engineering the RGB stuff as well... but this is where the Commander Pro gets complicated. In some of the RGB modes (certain types of fan/strip lighting) there seem to be quite heavy USB transmissions all the time. Maybe some of the fans or strips are "dumb" and thus the official Corsair LINK software transmits "change color and brigtness to <insert values>"  to the Commander Pro constantly. So I gave up on this... also terribly inefficient to waste CPU cycles and USB packets just to change colors.

So I only reverse engineered the "easy" RGB modes, where commands are only transmitted once during setup and then only when settings are actually changed by the user - the fan/strip handles the animation itself in this case.

I don't use RGB myself, so I think fan control, temperature measurement etc. is much more important and I could live without RGB support entirely. I think it is nice to have rather than need to have... but of course, this is my personal opinion. Always nice to have full support, but I hope the fans and sensors are first on the list, regarding the Commander Pro :-)

PostalBlab, 2018-04-10:
@pvh1987 is it possible to get in contact with you in private, so i can send you more information about the pro protocol?

audiohacked, 2018-04-11:
I think the H115i Pro and Commander Pro use the same protocol.... We should be able to expand the current support for both.

closed 2018-08-01 by audiohacked

Legion2, 2019-05-22:
@PostalBlab do you want to share some more information about the commands?
https://github.com/audiohacked/OpenCorsairLink/issues/70#issuecomment-380134782 these were already very helpful to build [an arduino library](https://github.com/Legion2/CorsairLightingProtocol), but there are two command which I have yet seen and do not know exactly what they are doing:
1. WriteLedRgbValue 0x31
2. WriteLedCount 0x3a

PostalBlab, 2019-06-01:
Sorry but i don't have more informations about these commands. Take a look at protocol/commanderpro/led.c , there are more infos then i have.

audiohacked, 2019-06-22:
@PostalBlab Again, thanks for documenting the Commander Pro. I had to reverse engineer command 0x29 (using SIV) and figured out the command format.

0x29 commands are fan get mode configuration commands. The request is of the form:
0x29 0x01 n (the rest of the bytes are zero)
where
'n' is the fan number (0-5 for fan 1-6)
The response is:
0x00 0x02 n
where
0x02 is the response size
'n' is the fan number (0-5 for fan 1-6)

audiohacked, 2019-06-22:
@PostalBlab If you could, please post any other information you may have.

Legion2, 2019-06-23:
@audiohacked I have have a working implementation of the Commander Pro as an Arduino library:
 - 0x00: https://github.com/Legion2/CorsairLightingProtocol/blob/dev/src/CorsairLightingFirmware.cpp
 - 0x10: https://github.com/Legion2/CorsairLightingProtocol/blob/dev/src/TemperatureController.cpp
 - 0x20: https://github.com/Legion2/CorsairLightingProtocol/blob/dev/src/FanController.cpp
 - 0x30: https://github.com/Legion2/CorsairLightingProtocol/blob/dev/src/LEDController.h

timlod, 2019-11-16:
Hi everyone,
I have a setup with the H100i GT V2 and the Commander PRO.
I would like to turn all LEDs off.
Is this possible, if so, what command does it? I have tried the usage instructions, but keep getting `Unsupported Generic LED Function`.
Thanks!