hyperkineticnerd/OpenCorsairLink #72: Works for HX1200i; update README

mmccool, 2018-01-29:
I have an HX1200i and a H115i.   I'd like to report that this seems to work for both.  Here is some sample output from my system.   Note the fan speed for the H115i is zero but that's because I don't have any fans plugged into the pump yet (I am using case fans, driven from the motherboard, to drive air through the radiator; I am *planning* to install the Corsair fans but need some stupid American (non-metric) bolts that I had to special order...).

My primary interest is in monitoring power consumption and controlling fans.  The numbers I am getting on power consumption are about what I expected.  

Anyhow, this note is not really an issue, other than perhaps to say the README should be updated.  While you are at it, a simple example of how to run the program and some expected output would be useful.

$ sudo ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: HX1200i!
Dev=1, CorsairLink Device Found: H115i!

Vendor: CORSAIR
Product: HX1200i
Firmware: NA
Temperature 0: 54.00 C
Temperature 1: 43.00 C
Powered: 245579 (2d.  20h)
Uptime: 79979 (0d.  22h)
Supply Voltage: 115.00
Total Watts: 304.00
Output 12v:
	Voltage 12.16
	Amps 22.50
	Watts 272.00
Output 5v:
	Voltage  4.97
	Amps  4.75
	Watts 23.00
Output 3.3v:
	Voltage  3.34
	Amps  1.88
	Watts  6.00

$ sudo ./OpenCorsairLink.elf --device 1
Dev=0, CorsairLink Device Found: HX1200i!
Dev=1, CorsairLink Device Found: H115i!

Vendor: Corsair
Product: H115i
Firmware: 2.8.0.0
Temperature 0: 33.60
Fan 0:	Mode 0x07
	Current/Max Speed 0/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 2850/0 RPM

mmccool, 2018-01-29:
Something fishy does seem to be going on with the H115i, though: why is the max speed 0 for both the fan and the pump?   Is the the TARGET max speed or the HISTORICAL max speed?   If the former, even the speed for the fan should not be zero.

mmccool, 2018-01-30:
I have now installed fans on the H115i and can confirm I am getting reasonable-looking "current" RPM readings.   The "max speed" is still getting reported as zero for both the pump and the fan.

sean808080, 2018-01-31:
Confirmed to work with HX1000i power supply.  Thanks!!

closed 2018-08-01 by audiohacked