hyperkineticnerd/OpenCorsairLink #79: H80iGT crashes when attempting fan control

bladesy, 2018-02-12:
Hello, I've been trying out this program with my H80iGT (I understand it doesn't seem to be supported, looking at the readme) but it seems to be compatible based on the following message:
`Dev=0, CorsairLink Device Found: Corsair Hydro Series 7289 USB Device!`
The first thing I tested was changing the LED colour, and it worked fine, as did changing the pump speed. Changing the pump speed or LED colour actually lead to the fan speeds being set to what I assume is the default of maximum performance, so it does seem like the program can affect the fan speeds of the H80iGT.
What didn't seem to work was manually changing the fan speeds - note that it gave back the correct current speeds and temperature, but the speed change doesn't seem to have been applied due to the crash:
[crash.log](https://github.com/audiohacked/OpenCorsairLink/files/1717354/crash.log)
I also noticed that the --fan argument takes 1 to 4, so I tried all of these values in turn to see the effect, and they all appeared to give the same crash.

audiohacked, 2018-03-25:
That's interesting. What options and how are you compiling the software?

audiohacked, 2018-03-25:
@bladesy Could you try changing the driver from corsairlink_driver_asetek to corsairlink_driver_hid ?

closed 2018-08-01 by audiohacked