hyperkineticnerd/OpenCorsairLink #80: H80i stack corruption?

grandinj, 2018-02-13:
with current master (just cloned now), I get the trace below.
Also, it seems to show rather random data from the fans?

(gdb) r --debug --dev 0
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/noel/OpenCorsairLink/OpenCorsairLink.elf --debug --dev 0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff70c6700 (LWP 2243)]
Checking USB device 0 (8087:8001)...
Checking USB device 1 (1d6b:0002)...
Checking USB device 2 (8087:8009)...
Checking USB device 3 (1d6b:0002)...
Checking USB device 4 (1d6b:0003)...
Checking USB device 5 (1b1c:0c04)...
Corsair product detected. Checking if device is H80i... Dev=0, CorsairLink Device Found: H80i!
Checking USB device 6 (045e:0039)...
Checking USB device 7 (0566:3008)...
Checking USB device 8 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H80i
Firmware: 16.5.0.0
Temperature 0: 35.39
Speed: 00 00
Max Speed: 00 00
Fan 0:	Not connected or failed
	Current/Max Speed 0/0 RPM
Speed: 00 00
Max Speed: 00 00
Fan 1:	Not connected or failed
	Current/Max Speed 0/0 RPM
Speed: 00 0D
Max Speed: 04 9F
Fan 2:	Not connected or failed
	Current/Max Speed 13/1183 RPM
Speed: 03 C4
Max Speed: 05 14
Fan 3:	Fixed RPM Mode (3PIN) set to 2000
	Current/Max Speed 964/1300 RPM
Speed: 00 00
Max Speed: 00 00
Pump:	Mode 0x0A
	Current/Max Speed 0/0 RPM
DEBUG: deinit done
*** stack smashing detected ***: /home/noel/OpenCorsairLink/OpenCorsairLink.elf terminated

Thread 1 "OpenCorsairLink" received signal SIGABRT, Aborted.
0x00007ffff7521428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt 10
#0  0x00007ffff7521428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff752302a in __GI_abort () at abort.c:89
#2  0x00007ffff75637ea in __libc_message (do_abort=do_abort@entry=1, fmt=fmt@entry=0x7ffff767b49f "*** %s ***: %s terminated\n") at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff760515c in __GI___fortify_fail (msg=<optimized out>, msg@entry=0x7ffff767b481 "stack smashing detected") at fortify_fail.c:37
#4  0x00007ffff7605100 in __stack_chk_fail () at stack_chk_fail.c:28
#5  0x0000000000401b1c in hydro_settings (scanned_device=..., settings=...) at main.c:221
#6  0x0000000000401cd9 in main (argc=0, argv=0x0) at main.c:252
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb)

crominger, 2018-03-19:
I was having the same issue until I booted into Windows and allowed Corsair Link to update the firmware to the latest. Now there is no stack error for me and everything seems to work fine.

audiohacked, 2018-03-25:
This looks to be either a Firmware problem, or a memory leak. As @crominger said, try updating the firmware.

audiohacked, 2018-03-25:
I just found out that a previous pull request was badly implemented. Line 175 of main.c re-utilized a previous variable that wasn't intended to be utilized as it was.

grandinj, 2018-03-27:
thanks for the updates. 

After basically banging various values at the cooler, I eventually managed to get it into a useful state :-)

Don't have windows on this box, so updating firmware unfortunately not an option

audiohacked, 2018-03-27:
You can try to run a Window Virtual Machine in VirtualBox and try updating that way.

closed 2018-08-01 by audiohacked