hyperkineticnerd/OpenCorsairLink #86: Is it to set the settings to be static?

p3x-robot, 2018-03-26:
Hi?
How are you?
I have a small question and issue. I tried to use it via cron, like this:
```bash
/home/patrikx3/Projects/patrikx3/home/linux-scripts/bin/OpenCorsairLink.elf --device 0 --pump-mode 5 --fan-mode 5
```
But the issue is that it is random, either sometimes is it full pumping and fan mode max, or it totally turned off.

How can I always keep it on max?

Right now, when I execute it, it is switching randomly what the pump and fan mode is.

The more issue is that, I am using the original Windows program as well, that is working fine.

So when I pump it on Linux, I have no idea which pump is on right now. I think you understand.

I checked the ```--help``` but it gives no info .

The pump is always Mode ```0x00```:
```text
Vendor: Corsair
Product: H80i V2
Firmware: 2.8.0.0
Temperature 0: 31.40
Fan 0:	Mode 0x07
	Current/Max Speed 2640/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 2910/0 RPM
```

I could check via ```bash``` if I can execute the pumping settings or keep the current, but there is no info.

I know it is open source and you have other thinks to do, so that's why just asking if there is a way to do this to make it statically fixed the settings instead of switching either random, or I don't my settings are correct.

Sincerely,
Patrik

p3x-robot, 2018-03-26:
If I execute only this:
```bash
./OpenCorsairLink.elf --device 0
```

This switches for sure every time or and off. That's all info I could collect.

PostalBlab, 2018-03-31:
Can you please set the pump mode under windows with CorsairLink (so you have a known state in the register), start GNU/Linux, run
`./OpenCorsairLink.elf --device 0` and paste the output here? I think the problem might be the wrong pump mode because OpenCorsairLink uses the same mode values for the fans and the pump, which is not correct for my H110i GT, see #91

p3x-robot, 2018-03-31:
[image]

[image]

[image]

Sending the settings form Linux, right away I got a chance, as we are playing Hearthstone...

p3x-robot, 2018-03-31:
Hey!
How are you?
Here is what you exactly said:

```text
patrikx3@workstation:~$ sudo -i
[sudo] password for patrikx3: 
root@workstation:~# /home/patrikx3/Projects/patrikx3/home/linux-scripts/bin/OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H80i V2!

Vendor: Corsair
Product: H80i V2
Firmware: 2.8.0.0
Temperature 0: 27.20
Fan 0:	Mode 0x07
	Current/Max Speed 1680/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 2880/0 RPM
root@workstation:~# 
```

p3x-robot, 2018-03-31:
When I execute again, it stops:
```text
root@workstation:~# /home/patrikx3/Projects/patrikx3/home/linux-scripts/bin/OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H80i V2!

Vendor: Corsair
Product: H80i V2
Firmware: 2.8.0.0
Temperature 0: 28.20
Fan 0:	Mode 0x07
	Current/Max Speed 2700/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 1920/0 RPM
root@workstation:~# 
```

p3x-robot, 2018-03-31:
Looks like the fan and the pump exchange/reverse.

p3x-robot, 2018-03-31:
Then 3rd and 4th and more is kind of random:
```text
root@workstation:~# /home/patrikx3/Projects/patrikx3/home/linux-scripts/bin/OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H80i V2!

Vendor: Corsair
Product: H80i V2
Firmware: 2.8.0.0
Temperature 0: 28.70
Fan 0:	Mode 0x07
	Current/Max Speed 1200/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 1890/0 RPM
root@workstation:~# /home/patrikx3/Projects/patrikx3/home/linux-scripts/bin/OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H80i V2!

Vendor: Corsair
Product: H80i V2
Firmware: 2.8.0.0
Temperature 0: 28.70
Fan 0:	Mode 0x07
	Current/Max Speed 1200/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 1890/0 RPM
root@workstation:~# /home/patrikx3/Projects/patrikx3/home/linux-scripts/bin/OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H80i V2!

Vendor: Corsair
Product: H80i V2
Firmware: 2.8.0.0
Temperature 0: 28.70
Fan 0:	Mode 0x07
	Current/Max Speed 1260/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 1980/0 RPM
root@workstation:~# /home/patrikx3/Projects/patrikx3/home/linux-scripts/bin/OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H80i V2!

Vendor: Corsair
Product: H80i V2
Firmware: 2.8.0.0
Temperature 0: 28.70
Fan 0:	Mode 0x07
	Current/Max Speed 2700/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 1950/0 RPM
```

p3x-robot, 2018-03-31:
The first one is that I want to make it always set it, but I can never know what is happening.

PostalBlab, 2018-03-31:
Didn't notice you use an H80i V2 which uses the asetek protocol. Don't know anything about it.

audiohacked, 2018-03-31:
For the H80i, the pump doen't have a Max mode, its either Quiet or Performance. For the fans, there's only two profiles; and no custom profile supported.

p3x-robot, 2018-03-31:
so for now just don't use it?

audiohacked, 2018-03-31:
No, its working! Performance, as you have selected, is going to be variable between 1850 and 1950 RPM. Fan mode set to Performance is going to be variable too. In other words, everything is work just like it is designed in the hardware.

p3x-robot, 2018-04-01:
hello! how are you?
so, it means, i just call it once on ```rc.local``` on boot, like this
```bash
./OpenCorsairLink.elf --device 0 --pump-mode 5 --fan-mode 5
```
and that's it?

audiohacked, 2018-04-01:
Yes

p3x-robot, 2018-04-01:
thanks so much! have good day!!! ciao!

closed 2018-08-01 by audiohacked