hyperkineticnerd/OpenCorsairLink #87: remove usage of size_t

audiohacked, 2018-03-28:
Go through the code and remove all usage of size_t. Size_t has an undefined bit-size. Depending on the context and use, size_t can be signed or unsigned or 8-, 16-, or 32-bits. There are some cases where you expect size_t to be uint32_t but is in fact int8_t.

PostalBlab, 2018-03-29:
size_t is defined as unsigned int with at least 16 bit, large enough to hold the size of the biggest supported object by the OS.
Imho sizeof() is exactly the use-case for size_t usage. snprintf uses size_t as well.
Why do you want to get rid of it in this case? The only questionable usage is in devic.c where it is used for counting.
`size_t corsairlink_device_count = sizeof(corsairlink_devices)/sizeof(corsairlink_devices[0]);`

audiohacked, 2018-03-29:
size_t is implementation-defined as "the unsigned integer type of the result of the sizeof operator". I want to constraint the code to the data ranges we should expect. As in the case of corsairlink_device_count, I don't see having more than 255 devices connected to a system. Unsigned integer type to me means, it can be 8-bits, 16-bits, 32-bit, or 64-bits depending on the compiler optimizations being performed.

closed 2018-08-01 by audiohacked