hyperkineticnerd/OpenCorsairLink #93: H115i Pro - LED color does not change

armanatz, 2018-04-08:
When running command `sudo OpenCorsairLink.elf --device 0 --led '{hexCode}'` where `{hexCode}` is something like `#ff0000` or `#00ffff`, the color of the LEDs don't actually change. On my device, they stay blue (`#0000ff`).

On an unrelated note, is there a way to change the LED mode from pulsing to static?

audiohacked, 2018-04-08:
The software framework for the H115i Pro LED system hasn't been implemented yet.

armanatz, 2018-04-08:
Oh my bad @audiohacked. Any chance of it being implemented or no dice?

audiohacked, 2018-04-10:
I plan on implementing the LED system when I get the time. I also need to implement the Commander Pro support too!

pvh1987, 2018-04-10:
I really look forward to support for the Commander Pro. I reverse engineered the protocol except most of the LED-related stuff and posted the details as an issue here on Github a while ago. Luckily the Commander Pro seems to remember the fan settings from my last boot into Windows, but it would be nice to be able to adjust the fans in Linux as well.

Also I have the H115i (not the Pro version, though) and OpenCorsairLink seems to be changing the fan speed even if I just run the program to get status. I hope this issue has been fixed - It wasn't a problem with an earlier version of OpenCorsairLink. Maybe I should try the newest version before I write this :-)

I hope that OpenCorsairLink will eventually turn into a daemon that prohibits direct communication with the devices but allows monitoring and changing settings through a small API. I think this would be much more secure, especially when using it in a multi-user environment. Also desktop/panel applets would be able to get temperature/fan info from the daemon instead of communicate with the hardware directly every time, for every user. However, I understand this work would be time consuming and basic hardware support is much more important at the moment.

Sorry for the thread hijacking... I just wanted to contribute with some ideas and thank you for all the good work so far :-)

audiohacked, 2018-04-28:
The 'Testing' Branch has support for the H115i Pro and H150i Pro LED Control!

closed 2018-08-01 by audiohacked