hyperkineticnerd/OpenCorsairLink #94: AX1500i supported?

Zerophase, 2018-04-15:
I'm able to provide information for the AX1500i. Just give me a sec to get back home and run lsusb. I can definitely help implement support for the device through the driver. Just want to avoid making the psu go pop in the process.

audiohacked, 2018-04-19:
Currently, No. The AXi PSUs utilize a PMBus which might never get supported. The AX1500i does have a built-in USB plug, but that might be going through a PMBUS-to-USB conversion. It would be pretty difficult for me to implement support for it without going out and buying one myself.

Zerophase, 2018-04-19:
I could help implement support. It's basically just monitoring the signals sent with wireshark? I just want to avoid damaging the unit in the process.

closed 2018-08-01 by audiohacked