hyperkineticnerd/OpenCorsairLink #95: H150i Pro

RobP76, 2018-04-17:
Hello could you tell me if the Corsair H150i Pro work please? 

Thanks

Rob

RobP76, 2018-04-17:
Does it work on the same driver?

From: Sean Nelson
Sent: 17 April 2018 15:19
To: audiohacked/OpenCorsairLink
Cc: RobP76; Author
Subject: Re: [audiohacked/OpenCorsairLink] H150i Pro (#95)

The H150i Pro should work, if a device entry is added to OpenCorsairLink! Currently, we have a device entry for the H115i Pro.

audiohacked, 2018-04-17:
Yes, the driver is the same for both H150i Pro and H115i Pro.

RobP76, 2018-04-17:
Ive only just got the cooler and I had to install windows to use it properly ill try Linux with your opencorsairlink and hope for the best

From: Sean Nelson
Sent: 17 April 2018 22:15
To: audiohacked/OpenCorsairLink
Cc: RobP76; Author
Subject: Re: [audiohacked/OpenCorsairLink] H150i Pro (#95)

Yes, the driver is the same for both H150i Pro and H115i Pro.

closed 2018-08-01 by audiohacked