hyperkineticnerd/OpenCorsairLink #96: LL120 fans

tgeorgiop, 2018-04-18:
Hello. Does this work with ll120 fans? Thanks.

audiohacked, 2018-04-18:
I don't know. Does it?

Belval, 2018-04-18:
What is your LL120 setup @tgeorgiop ? Are you using a commander pro, a lightning node pro, etc..?

Also what do you mean by "work" what do you want to achieve?

audiohacked, 2018-04-30:
I'm closing this since we haven't gotten any response in almost 2 weeks.

vlinkz, 2018-06-25:
Will it be possible to control (change led settings/fan speed) the LL120 fans with the Commander Pro any time soon? Currently, I can't make the Commander Pro do anything with OpenCorsairLink. Thanks.

Vagr9K, 2018-07-12:
Hey everyone!

I'm using LL120 with Lightning Node Pro, and apparently those fans don't have "hardware profiles" and get stuck on rainbow mode while on Linux.

I'd like to help you guys, but I'm not sure how.

vlinkz, 2018-07-12:
@Vagr9K, I believe that if you have the Lighting Node Pro you can use [Colorsair](https://github.com/Chlorek/Colorsair) to change the fan color/animations

Vagr9K, 2018-07-13:
Thanks @vlinkz!

shaneshort, 2018-08-07:
Hi Guys,
The LL Fans are no different than standard addressable lighting strips, it shouldn't be too difficult to figure the logic out.

benburkhart1, 2018-10-27:
Hey all, figured out this protocol for Lighting Node Pro + Corsair LL fans, I have the reversing information and a demo project available here:

https://github.com/benburkhart1/lighting-node-pro

closed 2019-10-03 by audiohacked