hyperkineticnerd/OpenCorsairLink #97: Cant build or not understanding how to run the app?

Rusherz, 2018-04-19:
Hi

I downloaded the folder as a zip and i dont have this file everyone is running OpenCorsairLink.elf

Unless they are using linux im on a mac i try to run ./Makefile both normally and with sudo but i get the same errors:

`clang: error: no such file or directory: '?='
clang: error: no such file or directory: 'gcc'
clang: error: no input files
./Makefile: line 20: CFLAGS: command not found
./Makefile: line 21: LDFLAGS: command not found
./Makefile: line 24: shell: command not found
./Makefile: line 24: CFLAGS: command not found
./Makefile: line 25: shell: command not found
./Makefile: line 25: LDFLAGS: command not found
./Makefile: line 27: OBJS: command not found
./Makefile: line 38: OBJS_LL: command not found
./Makefile: line 44: OBJS_PROTO: command not found
./Makefile: line 69: default:: command not found
./Makefile: line 71: all:: command not found
./Makefile: line 73: OBJS: command not found
./Makefile: line 73: OBJS_PROTO: command not found
./Makefile: line 73: OBJS_LL: command not found
./Makefile: line 73: OBJS: command not found
./Makefile: line 73: OpenCorsairLink.elf:: command not found
clang: error: no input files
./Makefile: line 74: CFLAGS: command not found
./Makefile: line 74: LDFLAGS: command not found
./Makefile: line 74: $^: command not found
./Makefile: line 76: fg: no job control
./Makefile: line 77: syntax error near unexpected token `newline'
./Makefile: line 77: `	$(CC) $(CFLAGS) -g -c -o $@ $<'`

audiohacked, 2018-04-20:
install Xcode

Rusherz, 2018-04-20:
I installed xcode and when I run ./Makefile or sudo ./Makefile i still get the same error as above is this just one of the "its just not going to work for you then" times

audiohacked, 2018-04-20:
Install the Xcode command line tools, and start installing additional libraries using OSX Homebrew (https://brew.sh/)

timodekruijf, 2018-07-31:
Hello, I have the same problem. Trying to understand what to do. Installed Xcode, Got Homebrew, managed to install for example "libusb". but have no clue what do do next, which libraries to install or how to compile and/or run this. Not a programmer but not a noob, either. Willing to learn.

audiohacked, 2018-07-31:
You don't execute the "Makefile". run make.

closed 2018-07-31 by audiohacked