hyperkineticnerd/OpenCorsairLink #98: Cant change any setting...

soofstad, 2018-04-20:
So it's kinda hard to find any concrete answer to how this work. So many issues are closed before any proper explanation are given.

So, can you realy only set any configuration once?!

I understand that a profile is written to the onboard storage of the pump (in my case H100i).
If there is no way to write again with OpenCorsaidLink, does anyone know how to "factory reset" the onboard storage so I'm able to run OpenCorsairLink a second time?

audiohacked, 2018-04-20:
Nothing is written to "onboard" storage. If you shut off the computer all your settings are lost! Please take the time to read the command help!

audiohacked, 2018-04-20:
It's not my problem if you can't take the time to figure out how the software works!

closed 2018-08-01 by audiohacked