hyperkineticnerd/OpenCorsairLink #99: RM750i PSU Kernel log flooding

kgara, 2018-04-23:
Hi. I would like to get rid of nasty kernel log flood while executing ocl for RM750i PSU's:
```
[Mon Apr 23 09:50:10 2018] hid-generic 0003:1B1C:1C0B.1D611: hiddev0,hidraw0: USB HID v1.11 Device [                                                ] on usb-0000:00:14.0-9/input0
[Mon Apr 23 09:50:10 2018] hid-generic 0003:1B1C:1C0B.1D612: hiddev0,hidraw1: USB HID v1.11 Device [                                                ] on usb-0000:00:14.0-8/input0
[Mon Apr 23 09:50:10 2018] hid-generic 0003:1B1C:1C0B.1D613: hiddev0,hidraw2: USB HID v1.11 Device [                                                ] on usb-0000:00:14.0-7/input0
[Mon Apr 23 09:50:10 2018] hid-generic 0003:1B1C:1C0B.1D614: hiddev0,hidraw3: USB HID v1.11 Device [                                                ] on usb-0000:00:14.0-5/input0
```
I made the script for telegraf metric collection (including aggregation of current watt consumption, will share it soon), it executes once every ten seconds for each PSU, so damn lot's of logs :). Any ideas what are they suppose to mean and how to turn them off? Maybe just some verbosity level :).

Other question not worth separate thread - is there any chances we may run ocl not as root? I made the workaround adding telegraf user to sudoers for my script only - not the best practice. Without it I get:
```
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/003: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
```
Thanks in advance!

audiohacked, 2018-04-23:
The kernel log flood must have something to do with libusb. I could be wrong but OCL doesn't interact with the kernel log. The libusb error has to do with udev/libusb/device permissions and is *kinda* outside the scope of OCL. You are more than welcome to submit a pull request with a fix! Save the fix in a 'udev' folder.

kgara, 2018-04-23:
I really doubt that this might be the permission issue - it occurs while running OCL from root or with sudo.  Anyway, thanks for advice - will try to google about this log in context of libusb.

audiohacked, 2018-04-23:
"hid-generic", "hiddev0", "hidraw3" and "USB HID v1.11" tell me its the kernel driver that's misbehaving. OCL uses libusb-1.0 which tells me it can be a funky interaction between the kernel and libusb. It **could** be a problem with OCL **IF** OCL was still using hidapi, but we're now using low-level libusb for every interaction with Corsair USB devices, most of which aren't "technically" HID devices. You'll have to figure out how to blacklist Corsair devices from the Linux Kernel's HID driver!

kgara, 2018-04-25:
OK. I created
`/etc/udev/rules.d/99-usbhid-corsair-psu.rules`
containing:
`SUBSYSTEM=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="1c0b", ATTR{authorized}="0"`
I reload rules
`udevadm control --reload-rules && udevadm trigger`
Stop getting this log message in dmesg. But now OCL does not provide any metrics.
```
# opencorsairlink --device 0
libusb: error [op_detach_kernel_driver] detach failed error -1 errno 113
Dev=0, CorsairLink Device Found: RM750i!

libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=3
...
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=3
Vendor: 
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=3
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=3
Product: 
Firmware: NA
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=3
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=3
Temperature 0:  0.00 C
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=3
...
```
Any ideas?

audiohacked, 2018-04-26:
https://stackoverflow.com/questions/38786343/prevent-usbhid-from-autoloading-when-usb-hid-device-is-plugged-in

kgara, 2018-04-26:
If you proposed this answer https://stackoverflow.com/a/39666866/6766656
Adding 
`DRIVERS=="usbhid", ACTION=="add"`
to 
`SUBSYSTEM=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="1c0b", ATTR{authorized}="0"`
together or each of them separate does nothing, as well as without idVendor and idProduct.
The second solution with usbhid module unloading obviously works (tested). But will I loose my usb keyboard and mouse? :) I am using ssh, the server, I am testing on stays far away - unable to check kb/mouse, but suppose I will loose it?

audiohacked, 2018-04-30:
I did some testing and research, the reason you see the log flood is from the hidraw driver being detached from the RM750i so that OCL can use the hardware. And then hidraw driver re-attaching itself back to the RM750i.

kgara, 2018-04-30:
And what might be the solution?

audiohacked, 2018-04-30:
Deal with it.

closed 2018-08-01 by audiohacked