hyperkineticnerd/OpenCorsairLink #108: some weird error

p3x-robot, 2018-05-10:
Hello!
How are you?
I am facing this error:

```text
patrikx3@workstation:/media/linux-nvme/home/patrikx3$ cd /home/patrikx3/Projects/OpenCorsairLink/
patrikx3@workstation:~/Projects/OpenCorsairLink$ git pull
Already up-to-date.
patrikx3@workstation:~/Projects/OpenCorsairLink$ make clean
rm -f OpenCorsairLink.elf main.o device.o driver.o print.o logic/options.o logic/options_fan.o logic/options_led.o logic/options_pump.o logic/scan.o logic/settings/commanderpro.o logic/settings/hydro_asetek.o logic/settings/hydro_asetekpro.o logic/settings/hydro_hid.o logic/settings/psu.o lowlevel/asetek.o lowlevel/commanderpro.o lowlevel/hid.o lowlevel/rmi.o protocol/asetek/core.o protocol/asetek/fan.o protocol/asetek/led.o protocol/asetek/pump.o protocol/asetek/temperature.o protocol/asetekpro/core.o protocol/asetekpro/fan.o protocol/asetekpro/led.o protocol/asetekpro/pump.o protocol/asetekpro/temperature.o protocol/commanderpro/core.o protocol/commanderpro/fan.o protocol/commanderpro/power.o protocol/commanderpro/temperature.o protocol/rmi/core.o protocol/rmi/power.o protocol/rmi/temperature.o protocol/rmi/time.o protocol/hid/core.o protocol/hid/fan.o protocol/hid/led.o protocol/hid/pump.o protocol/hid/temperature.o
patrikx3@workstation:~/Projects/OpenCorsairLink$ make
cc -std=c99 -pedantic-errors -Iinclude -I/usr/include/libusb-1.0 -g -c -o main.o main.c
cc -std=c99 -pedantic-errors -Iinclude -I/usr/include/libusb-1.0 -g -c -o device.o device.c
cc -std=c99 -pedantic-errors -Iinclude -I/usr/include/libusb-1.0 -g -c -o driver.o driver.c
cc -std=c99 -pedantic-errors -Iinclude -I/usr/include/libusb-1.0 -g -c -o print.o print.c
cc -std=c99 -pedantic-errors -Iinclude -I/usr/include/libusb-1.0 -g -c -o logic/options.o logic/options.c
cc -std=c99 -pedantic-errors -Iinclude -I/usr/include/libusb-1.0 -g -c -o logic/options_fan.o logic/options_fan.c
logic/options_fan.c: In function ‘fan_suboptions_parse’:
logic/options_fan.c:42:14: error: implicit declaration of function ‘getsubopt’ [-Wimplicit-function-declaration]
     switch ( getsubopt(&subopts, fan_options, &value) )
              ^
logic/options_fan.c:45:23: warning: format ‘%hhd’ expects argument of type ‘signed char *’, but argument 3 has type ‘uint8_t * {aka unsigned char *}’ [-Wformat=]
         sscanf(value, "%hhd", &settings->fan);
                       ^
logic/options_fan.c:50:23: warning: format ‘%hhd’ expects argument of type ‘signed char *’, but argument 3 has type ‘uint8_t * {aka unsigned char *}’ [-Wformat=]
         sscanf(value, "%hhd", &settings->fan_mode);
                       ^
Makefile:85: recipe for target 'logic/options_fan.o' failed
make: *** [logic/options_fan.o] Error 1
```

I am not a big problem.

Sincerely,
Patrik

audiohacked, 2018-05-10:
In the Makefile, comment out the line with"-pedantic-errors".

p3x-robot, 2018-05-10:
you mean i should create a pull request?
i am trying to use this repo and sync-it every day

p3x-robot, 2018-05-10:
ok, if someone needs it, here it is:

```bash
sed -i "s/CFLAGS += -pedantic-errors/#CFLAGS += -pedantic-errors/" Makefile
make clean
make
sed -i "s/#CFLAGS += -pedantic-errors/CFLAGS += -pedantic-errors/" Makefile
```

p3x-robot, 2018-05-18:
hello! how are you?
has the makefile changed?

closed 2018-08-01 by audiohacked