hyperkineticnerd/OpenCorsairLink #109: Corsair One Pro supported?

IanVermes, 2018-05-11:
I would like to get Linux to run on my Corsair One Pro but its dependent on the Corsair Link suite. I've read on the Corsair forums that its possible the Corsair One Pro has a different set of driver characteristics or possibly not (its largely speculation). I believe it uses similar components to those listed in your readme.

As the Corsair One/Corsair One Pro use mostly off the shelf Corsair cooling solutions, do you think OpenCorsairLink may support it?

Next time I'm with the machine I could open it up and try to report what cooling components it uses (can't seem to find details online).

Thanks for any advice.

audiohacked, 2018-05-11:
Try booting Linux and retrieve lsusb outputs: `lsusb` and `lsusb -d 1b1c:` and post them here!

IanVermes, 2018-05-12:
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1b1c:0c04 Corsair 
Bus 001 Device 003: ID 090c:1000 Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) Flash Drive
Bus 001 Device 005: ID 8087:0a2b Intel Corp. 
Bus 001 Device 002: ID 05ac:0267 Apple, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ lsusb -d 1b1c:
Bus 001 Device 004: ID 1b1c:0c04 Corsair 

$ lsusb -d 1b1c: -vBus 001 Device 004: ID 1b1c:0c04 Corsair 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c04 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      37
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

audiohacked, 2018-05-12:
Looks like the Corsair One should already be supported!

IanVermes, 2018-05-12:
That's great!

I went ahead and compiled it without changes to the makefile. I executed the elf and it just returned 0 exit status with no output at all.

Does it initialise some process in the background after executing the .elf? Is there some further documentation: I can see some of the options in include/logic/options.h.

choltz95, 2018-06-19:
@IanVermes any updates? Did you get this working?

audiohacked, 2018-08-02:
@choltz95 @IanVermes It's really hard for me to officially say the Corsair One Pro is supported without having the hardware myself. All I can really do is hope that someone who owns one is capable of adding support.

audiohacked, 2018-08-19:
@IanVermes What's the output of `sudo ./OpenCorsairLink.elf`? And what's the output of `sudo ./OpenCorsairLink.efl --dev 0`?

boyvanamstel, 2018-10-17:
I gave this a go on my Corsair ONE Pro. It doesn't seem to detect any devices.

```bash
$ sudo ./OpenCorsairLink.elf

$ sudo ./OpenCorsairLink.elf --dev 0

Detected 0 device(s), submitted device 0 is out of range
```

SBAM, 2018-11-04:
Greetings,

I added to device.c:
`    {`
`        .vendor_id = 0x1b1c,`
`        .product_id = 0x0c04,`
`        .device_id = 0x50,`
`        .name = "Corsair One",`
`        .read_endpoint = 0x01|LIBUSB_ENDPOINT_IN,`
`        .write_endpoint =`
`            LIBUSB_REQUEST_TYPE_CLASS | LIBUSB_RECIPIENT_INTERFACE | LIBUSB_ENDPOINT_OUT,`
`        .driver = &corsairlink_driver_coolit,`
`        .led_control_count = 0,`
`        .fan_control_count = 1,`
`        .pump_index = 0,`
`    },`

Then I can run:
_sudo ./OpenCorsairLink.elf --device 0 --fan channel=0,mode=5_
  _Dev=0, CorsairLink Device Found: Corsair One!_

  _Vendor: Corsair_
  _Product: Corsair One_
  _Firmware: 1.0.14_
  _Temperature 0: 26.94 C_
  _Fan 0:	Default Mode (4PIN)_
  _Current/Max Speed 419/2800 RPM_
  _Fan 1:	Default Mode (4PIN)_
  _Current/Max Speed 0/2800 RPM_
  _Pump:	Mode 0x87_
  _Current/Max Speed 419/2800 RPM_

This has however no effect on fan speed. As fas as I understand, according to **corsair link**, there is only one fan that can be controlled, the top case fan.
Pump mode and second fan cannot be controlled from **corsair link**.

This is irrelevant to me but the front case leds intensity should be controllable also.

A while ago (on commit 0be1661135fc188be724bd214ca1d2d529c5923b ), I could set fan speed to max using this command line (and using old corsairlink_driver_hid ):
`while [ true ]; do`
`    ./OpenCorsairLink.elf --device 0 --fan 1 --fan-rpm 255`
`    sleep 10`
`done`

I had to keep it in a loop because it would revert to default settings after 15-20sec.

I hope this helps, cheers !

IanVermes, 2018-11-12:
@boyvanamstel Sorry I am unable to check as the Corsair One is being repaired by Corsair (bad PCIE riser I believe), I'll try it when its back.

IncludedInside, 2019-06-16:
I confirm the findings of SBAM. I am running Ubuntu 18.04.2 LTS
If I use git and branch off from the 25th of Feb, apply his device.c changes, I get a build where
./OpenCorsairLink.elf --device 0 --fan 1 --fan-rpm 255 
speeds up the fan for a short period of time. Running that command repeatedly stops the computer overheating under full load.

It also causes this to appear in the system log each time its executed:

Jun 16 18:20:46 Corsair upowerd[1505]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1B1C:0C04.0042
Jun 16 18:20:46 Corsair upowerd[1505]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0
Jun 16 18:20:50 Corsair upowerd[1505]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1B1C:0C04.0042
Jun 16 18:20:50 Corsair upowerd[1505]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0
Jun 16 18:20:50 Corsair upowerd[1505]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0
Jun 16 18:20:50 Corsair upowerd[1505]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0
Jun 16 18:20:50 Corsair kernel: [14775.004487] hid-generic 0003:1B1C:0C04.0043: hiddev1,hidraw2: USB HID v1.11 Device [Corsair Memory, Inc. Integrated USB Bridge] on 

Applying the device.c change to the latest release gives a build that I have not been able to get working. Specifically

sudo ./OpenCorsairLink.elf --device 0 --fan channel=1,mode=1,rpm=1000
reports
Vendor: Corsair
Product: Corsair One
Firmware: 1.0.14
Temperature 0: 26.53 C
Fan 0:	Default Mode (4PIN)
	Current/Max Speed 416/2800 RPM
Fan 1:	Default Mode (4PIN)
	Current/Max Speed 0/2800 RPM
Pump:	Mode 0x87
	Current/Max Speed 416/2800 RPM
Unsupported Fan Mode

and so does 
sudo ./OpenCorsairLink.elf --device 0 --fan channel=1,mode=0,pwm=100

These also create the same syslog messages.

So the old version is encouraging. If there is a way of preventing the speed being reset, running the OpenCorsairLink command once before starting any heavy CPU activity would be fine. Running it in a loop generates too much junk in the system log to be a practical way forward.

I am happy to make any changes you suggest to the code and report back if that helps.

audiohacked, 2019-06-19:
@IncludedInside I think I see the problem. Under Ubuntu, you have a service called Upower running, which is taking control of the corsair fans.

Beyond that, I can't really help.

IncludedInside, 2019-06-20:
So the correct approach would be a fix for upower to get it to set the fan speed correctly.

I do not need to learn how to drive that now because I found much simpler solution. If you detach pin 4 on the 4 wire connector to the top of case fan, it runs at full speed all the time. Nothing software can do to get in the way now :-)

Sure it's noisier than strictly necessary, but not offensively so. 

The fan connector pinout is 
0 = GND
1 = 12V 
2 = TACH - a pulse stream that tells the motherboard the current fan speed
3 = PWM - an analog speed control signal. If absent the fan runs full speed.

This guy works it out
https://www.youtube.com/watch?v=gKHww3qJbs8

I am just going to put this down to a learning exercise and be more careful what hardware I buy in future. Avoiding Corsair being a good starting point.

expipiplus1, 2020-02-09:
I was unable to get @SBAM's patch working (I believe we have different machines, as my USB product ID is different). For my corsair one with `1b1c:0c14` (same as the "H110 Pro") I've made a little utility which prints the fan/pump speeds and coolant temps: https://github.com/expipiplus1/corsair-one-stats. Using this tool one can observe that the fan/pump speeds do respond on Linux to the system temperature (I still need to check if this is true for the GPU temperature too).

expipiplus1, 2020-02-09:
Oops, I bricked my fan controller :/

SBAM, 2020-05-06:
> I was unable to get @SBAM's patch working (I believe we have different machines, as my USB product ID is different). For my corsair one with `1b1c:0c14` (same as the "H110 Pro") I've made a little [...]

Dear all, here is a [small daemon](https://github.com/SBAM/corsair_one_d) that watches coretemps/coolant_temp and increases top fan speed if thresholds are exceeded. @expipiplus1 it's only designed to manage the older product **0c04**.
Thanks for your research, I might check how to interact with pumps now (I also noted that your hardware seems to have 2 separate liquid circuits, while mine only has 1).