hyperkineticnerd/OpenCorsairLink #111: Custom curves not supported?

DanySK, 2018-05-15:
Hi,
I am experimenting in the master branch, and I believe the manual control of fan speed over temperatures is not working. In fact, seems to me that `SUBOPTION_FAN_TEMPS` and `SUBOPTION_FAN_SPEEDS` are entirely ignored in `options_fan.c`.

Executing:
`sudo ./OpenCorsairLink.elf --device 0 --pump mode=3 --fan mode=6,temps="0:20:25:30:35:40",speeds="0:10:30:60:80:100"`

causes:
```
Unknown suboption `0:20:25:30:35:40'
Unknown suboption `0:10:30:60:80:100'
```

Are custom curves a work in progress?

chenyangh, 2018-05-16:
Yes, I found the same. And it was supported in an older version.

audiohacked, 2018-05-16:
@DanySK Yes, custom curves are a work in progress. Support was dropped due to the rewrite of the command line option parser.

audiohacked, 2018-05-16:
By the way, you should be on Testing Branch.

DanySK, 2018-05-16:
Switched branch. I still see inconsistencies:
```
Fan:
        --fan channel=N,mode=N,pwm=PWM,rpm=RPM,temps=TEMP:TEMP:TEMP,speeds=SPEED:SPEED:SPEED
```
but right below:
```
temps <CSV of Temperatures>     :Define Comma Separated Values of Temperatures for Fan.
                speeds <CSV of Speed Percentage>        :Define Comma Separated Values of RPM Percentage for Fan.
```
Which syntax would you like to support? I believe that with the current `getsubopt` parsing it'd be way easier to support a colon separator.

DanySK, 2018-05-16:
I'd like some clarifications about `struct option_parse_return`. It is unclear to me what `struct fan_table fan1` and `struct fan_table pump` represent, as they seem to me redundant w.r.t. `struct fan_control fan_ctrl` and `struct pump_control pump_ctr`. The latter two are currently unused: are them a leftover from a refactoring, or are them supposed to replace `fan1` and `pump`?

audiohacked, 2018-05-16:
`struct fan_control` and `struct pump_control` are intended to replace all the other variables in `struct option_parse_return`

DanySK, 2018-05-17:
Closed means that the support is now there, that you are working on it but not accepting contributions, that they won't be supported, or what?

audiohacked, 2018-05-17:
Working on it, but if you provide a pull request that includes support I'll take it.

DanySK, 2018-05-18:
Sure, I'm willing to help, I need your guidance though as I'm not expert of the Corsair AIOs internals, and I can only test on my own H100i.
I could start with parsing the subcommand for temps and speeds and populating `struct option_parse_return`. If you agree, I'd use the new `fan_ctrl` and `pump_ctrl`.

closed 2018-08-01 by audiohacked

notlesh, 2019-10-19:
Was there a commit where this was actually working?

audiohacked, 2019-10-19:
The main problem some devices support custom fan curves and other devices do **not** support custom fan curves. The support depends on the device in question.

notlesh, 2019-10-20:
Oh, I see. I have a `h110i`, which I thought supported them, but I can't seem to do that under Windows at the moment.

Assuming I had a device that supported custom curves, does the current or any past version of `OpenCorsairLink` support this?

audiohacked, 2019-10-21:
@notlesh If you have a device that supports it, then any version of OpenCorsairLink that supports that device will have support for custom curves.