hyperkineticnerd/OpenCorsairLink #113: Weird behavior when running the same command twice and settings change when just querying device information (H115i)

pvh1987, 2018-05-16:
I believe that the issue "Inconsistent behavior launching the same command multiple times" by DanySK has been closed too early. Here is my investigation of the similar problems I have with the H115i and OpenCorsairLink:

I have noticed that just querying information about a device can actually change settings. For example, when I just run OpenCorsairLink, it outputs the following:

    $ ./OpenCorsairLink.elf 
    Dev=0, CorsairLink Device Found: HX850i!
    Dev=1, CorsairLink Device Found: H115i!

    Which is correct. Then I want more information about the H115i and I run

    $ ./OpenCorsairLink.elf --device 1
    Dev=0, CorsairLink Device Found: HX850i!
    Dev=1, CorsairLink Device Found: H115i!

    Vendor: Corsair
    Product: H115i
    Firmware: 2.8.0.0
    Temperature 0: 33.70
    Fan 0:	Mode 0x07
	    Current/Max Speed 720/0 RPM
    Pump:	Mode 0x00
	    Current/Max Speed 2820/0 RPM

Now I can clearly hear the fan is spinning faster. And I have not changed any parameter at all. Running the command again, the fan become quiet again:

    $ ./OpenCorsairLink.elf --device 1
    Dev=0, CorsairLink Device Found: HX850i!
    Dev=1, CorsairLink Device Found: H115i!

    Vendor: Corsair
    Product: H115i
    Firmware: 2.8.0.0
    Temperature 0: 33.70
    Fan 0:	Mode 0x07
	    Current/Max Speed 660/0 RPM
    Pump:	Mode 0x00
	    Current/Max Speed 1860/0 RPM

    Running the command a third time speeds up the fan again:

    $ ./OpenCorsairLink.elf --device 1
    Dev=0, CorsairLink Device Found: HX850i!
    Dev=1, CorsairLink Device Found: H115i!

    Vendor: Corsair
    Product: H115i
    Firmware: 2.8.0.0
    Temperature 0: 33.70
    Fan 0:	Mode 0x07
	    Current/Max Speed 1620/0 RPM
    Pump:	Mode 0x00
	    Current/Max Speed 1920/0 RPM

And so it goes on indefinitely - at least, that is my experience.

I am not sure if there is a particular pattern to the fan speed change each time. However, the change seems to be applied after the output is printed, so the last run of the command shows the speed that the previous command applied - but I am not totally sure on this.

Running the command several times gives me fan speeds (in RPM) of 660, 660, 780, 720, 2100, 2040, 1080, 1380, 900, 1080, 840, 720, 660, ...

So, something weird is going on.

Next, I tried to actually change the fan speed setting myself. It appears in the --help information that I have to specify fan number 1 in the command line argument although the fan is printed as number 0:

    $ ./OpenCorsairLink.elf --device 1 --fan 1 --fan-mode 0 --fan-pwm 100
    Dev=0, CorsairLink Device Found: HX850i!
    Dev=1, CorsairLink Device Found: H115i!

    Vendor: Corsair
    Product: H115i
    Firmware: 2.8.0.0
    Temperature 0: 33.80
    Fan 0:	Mode 0x07
	    Current/Max Speed 660/0 RPM
    Pump:	Mode 0x00
	    Current/Max Speed 1950/0 RPM

Nothing happened... Running it again, a can hear a faint speed change:

    $. /OpenCorsairLink.elf --device 1 --fan 1 --fan-mode 0 --fan-pwm 100
    Dev=0, CorsairLink Device Found: HX850i!
    Dev=1, CorsairLink Device Found: H115i!

    Vendor: Corsair
    Product: H115i
    Firmware: 2.8.0.0
    Temperature 0: 33.80
    Fan 0:	Mode 0x07
	    Current/Max Speed 600/0 RPM
    Pump:	Mode 0x00
	    Current/Max Speed 1950/0 RPM

The third time, I can hear the fan is turning on and off periodically before becoming quiet. Then after quite some time, the fan speeds up to full speed and stays there. Very strange.

    $ ./OpenCorsairLink.elf --device 1 --fan 1 --fan-mode 0 --fan-pwm 100
    Dev=0, CorsairLink Device Found: HX850i!
    Dev=1, CorsairLink Device Found: H115i!

    Vendor: Corsair
    Product: H115i
    Firmware: 2.8.0.0
    Temperature 0: 33.80
    Fan 0:	Mode 0x07
	    Current/Max Speed 1560/0 RPM
    Pump:	Mode 0x00
	    Current/Max Speed 1920/0 RPM

After somehow getting the fan silent again, I tried to change the pump setting:

    $ ./OpenCorsairLink.elf --device 1 --pump-mode 3
    Dev=0, CorsairLink Device Found: HX850i!
    Dev=1, CorsairLink Device Found: H115i!

    Vendor: Corsair
    Product: H115i
    Firmware: 2.8.0.0
    Temperature 0: 33.70
    Fan 0:	Mode 0x07
	    Current/Max Speed 660/0 RPM
    Pump:	Mode 0x00
	    Current/Max Speed 1920/0 RPM

It made the fan speed up but apparently no pump mode change. Running the same command several times changes the fan speed again each time. The same thing goes on if I set pump-mode to 5:

    $ ./OpenCorsairLink.elf --device 1 --pump-mode 5
    Dev=0, CorsairLink Device Found: HX850i!
    Dev=1, CorsairLink Device Found: H115i!

    Vendor: Corsair
    Product: H115i
    Firmware: 2.8.0.0
    Temperature 0: 33.30
    Fan 0:	Mode 0x07
	    Current/Max Speed 1380/0 RPM
    Pump:	Mode 0x00
	    Current/Max Speed 2910/0 RPM

I usually run OpenCorsairLink like this:

    $ ./OpenCorsairLink.elf --device 1 --pump-mode 3 --fan 1 --fan-mode 6 --fan-temps 20,59,60,69,70,100 --fan-speeds 30,30,60,60,100,100
    Dev=0, CorsairLink Device Found: HX850i!
    Dev=1, CorsairLink Device Found: H115i!

    Vendor: Corsair
    Product: H115i
    Firmware: 2.8.0.0
    Temperature 0: 33.50
    Fan 0:	Mode 0x07
	    Current/Max Speed 660/0 RPM
    Pump:	Mode 0x00
	    Current/Max Speed 1950/0 RPM

It makes the system quiet and nothing weird happens, even if I run the same command several times.

However, there is still a problem. Notice that in all the outputs above, the fan mode stays at 0x07 even switching from fixed PWM to custom curve. The pump mode is also stuck at 0x00. Furthermore, is there any correlation between the modes printed in the output as hex values and the modes set by command line arguments?

Last, but not least, I am unsure if the custom curve is actually working like it should. Yesterday I was compiling a lot of stuff (I am a Gentoo user) and the CPU itself reported temperatures well over 60 degrees Celcius according to the "sensors" command. I could not hear the fan change at all, even though it was set to go to 60% PWM when the H115i measures >= 60 degrees Celcius.

I investigated this further, and found out that the H115i never reported more than 40 degrees Celcius. This could be because the CPU reports peak temperatures (not sure on this), measured "inside the cores" and the temperatures was fluctuating quite a bit. On the other hand, the H115i measures the temperature on the outside of the CPU, through the thermal paste and so on, which effectively acts as a low-pass filter, smoothing out the measurements and resulting in an averaged temperature measurement. This could explain why the H115i never reported more than 40 degrees and thus why the fan speed never changed. Thus, the custom curve might actually be working. I will have to investigate this further with stress tests and compare Windows vs Linux results.

So, my conclusion right now is that OpenCorsairLink is not applying settings correctly and apparently changes settings even if no parameters are changed at the command line. At least this is my personal experience with the H115i in OpenCorsairLink. I have not played around with my PSU yet, but it does not seem to change anything when I just query information - so I think the problems are related to the H115i only.

I got a second computer with a H100i v2 that I am currently installing Linux on. I plan to eventually do some testing here as well, to find out if the problems are only related to the H115i or the H100i v2 as well.

By the way, all this testing was done with the current master branch. I have noted similar inconsistensies between command line arguments and the actual hardware settings in previous versions of OpenCorsairLink but this is the first time I actually did a deep investigation of the problem. I find it especially weird that the fan speed changes even when just querying information, but it is also strange that reported fan and pump mode does not seem to be related to the given command line arguments.

pvh1987, 2018-05-17:
I wonder why this issue is closed already and with no comments at all?

I did try the testing branch today and the problem is still there. The fans are still changing speed for no reason.

Furthermore, my H115i is now a H110i GT V2 according to the testing branch, which is incorrect:

    ./OpenCorsairLink.elf 
    Dev=0, CorsairLink Device Found: Commander PRO!
    Dev=1, CorsairLink Device Found: HX850i!
    Dev=2, CorsairLink Device Found: H110i GT V2!

However, I like that the Commander PRO shows up on the list :-)

@audiohacked I understand if you are busy and don't have time to fix this at the moment, but closing an issue before it is solved is probably not the best idea? I certainly don't expect an issue to be fixed within days, weeks, months or maybe even at all. I just want to help the development of OpenCorsairLink :-)

pvh1987, 2018-05-17:
@audiohacked Thanks for your response :-) Interesting - I wasn't aware of that. I guess Corsair either renamed the product or reused USB IDs from the H115i to the H110i GT V2. If so, I apologize for saying it was incorrect. At least it's Corsair's fault, not our fault :-)

I will definitely consider contributing code as well. However, I think testing is also part of the development cycle. I didn't mean to be rude by opening this issue - I just wanted to help fix a problem. I still appreciate all the work done so far :-)

I think I might have the time in a few days to take a look at the code and see if I can find a solution myself.

bluntelk, 2018-07-07:
So I went digging and I think I have identified the problem.

In protocol/asetek/fan.c we have the function "corsairlink_asetek_fan_mode()". This gets called from the settings function "hydro_asetek_settings" via "dev->driver->fan.profile()"

"corsairlink_asetek_fan_mode()" gets called with the fan mode parameter "UNDEFINED". The "struct fan_table curve;" variable is then defined, but no value set - which if I remember my C correctly will cause it to use undefined memory.

"corsairlink_asetek_fan_curve()" is then invoked with this undefined command curve. which sometimes has the right values to kick the fan speeds up and down.

In short - "corsairlink_asetek_fan_curve()" is writing random values to the device

felixsanz, 2018-07-24:
I'm also experiencing random fan speeds when just running the command with no settings

Flova, 2018-07-29:
I just downloaded this library and have the same issue on my H100i V2. If I run the following command every thing shows up correctly, but the fan speeds are switching in the described pattern.
```
sudo OpenCorsairLink --device 0
Dev=0, CorsairLink Device Found: H100i V2!

Vendor: Corsair
Product: H100i V2
Firmware: 2.8.0.0
Temperature 0: 37.10 C
Temperature 1: 37.10 C
Temperature 2: 37.10 C
Fan 0:	Mode 0x07
	Current/Max Speed 1380/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 1950/0 RPM
```

felixsanz, 2018-07-31:
@audiohacked any update on this please?

audiohacked, 2018-07-31:
I'm busy with work, so I haven't dug into why.

felixsanz, 2018-08-02:
@audiohacked You've made an awesome effort creating this library, hope you find the time to keep it stable. thanks!

audiohacked, 2018-08-05:
I'm closing this for three reasons:
 - the UI has changed.
 - major changes have been made in the logic to hopefully fix the query/reading of settings.
 - the API for H115i Pro seems incomplete for some reason.

closed 2018-08-05 by audiohacked