hyperkineticnerd/OpenCorsairLink #114: --pump-mode isn't recognized

jchevarley, 2018-05-17:
Attempting to run:

sudo OpenCorsairLink --device 0 --pump-mode 3 --fan-temps 0,20,25,30,35,40 --fan-speeds 0,10,30,60,80,100

I get: 

OpenCorsairLink: unrecognized option '--pump-mode'

Most other variations result in the same,  but:

sudo OpenCorsairLink --device 0 --pump mode = 3 --fan-temps 0,20,25,30,35,40 --fan-speeds 0,10,30,60,80,100

Produces a seg fault. The CLI help documentation has '--pump mode=<mode>' - not sure if the hyphen was dropped and broke parsing?

bakgwailo, 2018-05-25:
OK - what is the proper syntax? I have tried different permutations for --help and still can't quite get it right to work. Apologize for the stupid question.

bakgwailo, 2018-05-25:
Never mind, I am an idiot, this works:

sudo OpenCorsairLink --device 0 --pump mode=3 --fan temps=0:20:25:30:35:40 --fan speeds=0:10:30:60:80:100

closed 2018-08-01 by audiohacked