hyperkineticnerd/OpenCorsairLink #118: H115i Pro RGB fan speed not changing

mbahri, 2018-07-15:
Hello

I have a Corsair H115i Pro RGB and I'm trying to control fan speeds from OpenCorsairLink but whatever I try doesn't change anything.

Example:

```
sudo ./OpenCorsairLink.elf --device 0 --fan channel=4,mode=5
Dev=0, CorsairLink Device Found: H115i Pro!

Vendor: Corsair
Product: H115i Pro
Firmware: 1.0.4.0
Temperature 0: 35.00 C
Fan 0:	Mode 0x07
	Current/Max Speed 820/0 RPM
Fan 1:	Mode 0x07
	Current/Max Speed 810/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 2130/0 RPM
```
Should make the fan spin faster.

```
sudo ./OpenCorsairLink.elf --device 0 --fan channel=1,mode=0,pwm=100
Unknown suboption `100'
Dev=0, CorsairLink Device Found: H115i Pro!

Vendor: Corsair
Product: H115i Pro
Firmware: 1.0.4.0
Temperature 0: 35.09 C
Fan 0:	Mode 0x07
	Current/Max Speed 821/0 RPM
Fan 1:	Mode 0x07
	Current/Max Speed 809/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 2130/0 RPM
```
The pwm suboption failed.

```
sudo ./OpenCorsairLink.elf --device 0 --fan channel=4,mode=6,temps=25:30:35,speeds=50:75:100
Unknown suboption `25:30:35'
Unknown suboption `50:75:100'
Dev=0, CorsairLink Device Found: H115i Pro!

Vendor: Corsair
Product: H115i Pro
Firmware: 1.0.4.0
Temperature 0: 35.00 C
Fan 0:	Mode 0x07
	Current/Max Speed 823/0 RPM
Fan 1:	Mode 0x07
	Current/Max Speed 815/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 2160/0 RPM
```
Same as above.

Am I doing things wrong or are these features not supported yet?

mbahri, 2018-07-15:
I compiled from branch testing.

audiohacked, 2018-07-16:
The code for options pwm, temps, and speeds hasn't been committed yet.

mbahri, 2018-07-16:
OK, and what about predefined modes such as performance or quiet?

In case that's useful, my cooler is currently operating on a custom mode made in corsair link : the pump is in balanced mode and the fans have a custom curve.

audiohacked, 2018-07-16:
If you find anything that doesn't work, please consider contributing.

closed 2018-08-05 by audiohacked