hyperkineticnerd/OpenCorsairLink #124: Unit Tests

audiohacked, 2018-08-06:
**Is your feature request related to a problem? Please describe.**
I'm always wondering if the OpenCorsairLink code logic is correct.

**Describe the solution you'd like**
I'd like as much unit test coverage as possible. Considering OpenCorsairLink deals with actual hardware, we would need mock test to test all of the hardware variants.

**Describe alternatives you've considered**
None

**Additional context**
N/A

ZirconiumX, 2018-08-18:
Hi! I'm from the [ckb-next](https://github.com/ckb-next) team, and we had a similar problem on our end. Our solution was to hack a USBIP driver into submission, and we got the mess that is [FaKeyboard](https://github.com/ckb-next/FaKeyboard).

Given we're both working on Corsair stuff, it'd be nice if you visited our IRC (Freenode #ckb-next).

audiohacked, 2018-08-20:
I found a set of articles on how to unit test software that interacts with hardware. 
http://blog.wingman-sw.com/archives/category/mocking-hardware
http://blog.wingman-sw.com/archives/7#more-7
http://blog.wingman-sw.com/archives/8#more-8
http://blog.wingman-sw.com/archives/507#more-507

audiohacked, 2018-08-20:
Here's an article on using CMocka: https://lwn.net/Articles/558106/