hyperkineticnerd/OpenCorsairLink #126: Commander Pro with LL120's & H150i Pump Mode

theupriser, 2018-08-17:
**Is your feature request related to a problem? Please describe.**
First, I've got a Commander PRo with 6 LL120's and I can't send commands to them. They give back Unsupported Generic Fan Function & Unsupported Generic LED Function. Also the temperatures and voltages on the device are 0.00. also the firmware is not recognized. 

Second the fans on the H150i are controllable but I can't set the pump mode. It gives back Unsupported Generic Pump Function. 

**Describe the solution you'd like**
So what I like to see is that all of these functions work. The --led and --fans on the Commander Pro and the --pump mode on the H150i.

**Describe alternatives you've considered**
I've considered contributing to the project for these devices since i'm a programmer. I'm not really into C++ but that could change :). I'd like to have some advice on where to start and how to do it. Maybe someone has already done these "fixes" but i sure like to help!

audiohacked, 2018-08-18:
The Commander Pro and the H150i are the newest devices with on-going work being done, and therefore don't have full support for all device features.

ekpen, 2018-10-29:
Greetings:

In lieu of LL120s fan canMLl120/140mm fans work without issues ?

thanks.

George

Mipirakas, 2018-11-11:
theupriser
For me, it is the other way around. I have a h150i pro and i can control and set the pump mode but i can't control my fans. i have replaced the stock fans with some noctuas though but on windows this combo works great.

Mipirakas, 2018-11-11:
I connected my stock fans and did a git reset to the date of your comment so that i would have the same version as you had but i still wasn't able to control my fan speeds. what is the exact procedure you followed to install this software and what is your command to control the fan speed? thank you.

truatpasteurdotfr, 2019-07-05:
setting fans at 50%:
` for i in `seq 0 4`; do sudo ./OpenCorsairLink.elf --device 0 --fan channel=$i,mode=0,pwm=50 ; done`

```
[truadm@ld18-1012 OpenCorsairLink]$ sudo ./OpenCorsairLink.elf --device 0 
Dev=0, CorsairLink Device Found: Commander PRO!

Vendor: Corsair
Product: Commander PRO
Firmware: V0.8.210
Temperature 0: 34.55 C
Temperature 1: 33.90 C
Temperature 2: 32.05 C
Temperature 3: 31.60 C
Output 12v: 11.89 V
Output 5v:  4.88 V
Output 3.3v:  3.33 V
Fan 0:  Mode: Auto/Disconnected
        PWM: 50%
        RPM: 1140
Fan 1:  Mode: Auto/Disconnected
        PWM: 50%
        RPM: 1139
Fan 2:  Mode: Auto/Disconnected
        PWM: 50%
        RPM: 1081
Fan 3:  Mode: Auto/Disconnected
        PWM: 50%
        RPM: 1330
Fan 4:  Mode: Auto/Disconnected
        PWM: 50%
        RPM: 1110
Fan 5:  Mode: Auto/Disconnected
        PWM: 0%
        RPM: 0

```

fans at 500 rpm:
`for i in `seq 0 4`; do sudo ./OpenCorsairLink.elf --device 0 --fan channel=$i,mode=1,rpm=500 ; done`

```
[truadm@ld18-1012 OpenCorsairLink]$ sudo ./OpenCorsairLink.elf --device 0 
Dev=0, CorsairLink Device Found: Commander PRO!

Vendor: Corsair
Product: Commander PRO
Firmware: V0.8.210
Temperature 0: 34.54 C
Temperature 1: 33.88 C
Temperature 2: 32.11 C
Temperature 3: 31.60 C
Output 12v: 11.89 V
Output 5v:  4.89 V
Output 3.3v:  3.33 V
Fan 0:  Mode: Auto/Disconnected
        PWM: 0%
        RPM: 513
Fan 1:  Mode: Auto/Disconnected
        PWM: 0%
        RPM: 618
Fan 2:  Mode: Auto/Disconnected
        PWM: 0%
        RPM: 475
Fan 3:  Mode: Auto/Disconnected
        PWM: 0%
        RPM: 488
Fan 4:  Mode: Auto/Disconnected
        PWM: 0%
        RPM: 497
Fan 5:  Mode: Auto/Disconnected
        PWM: 0%
        RPM: 0

```

audiohacked, 2019-10-03:
Closing this until I can get hardware to test with.

closed 2019-10-03 by audiohacked