hyperkineticnerd/OpenCorsairLink #127: Not working reliably on h100i v2

farnoy, 2018-08-17:
**Describe the bug**
I can't set any fan or pump mode/rpm reliably. It works every 2nd or 3rd invocation, and it doesn't seem to change the profile, just sets constant RPM based on god-knows-what.

For example, my goto command is this:

`$ sudo OpenCorsairLink --device 0 --pump mode=5 --fan channel=0,mode=4`

but if I invoke `$ sudo OpenCorsairLink --device 0` to see my changes, I see that no mode has been changed, sometimes just the RPM changes between absolute-minimum and absolute-maximum. I can't seem to control this at all.

**To Reproduce**
Steps to reproduce the behavior:
1. `$ sudo OpenCorsairLink --device 0 --pump mode=5 --fan channel=0,mode=4`
2. Observe the new settings (either 800RPM or 2.8k), mode does not change at all. Also notice, that merely checking the status of the device changes fan RPMs.
3. Either wait and watch it continue forever, or go to step 1

**Expected behavior**
I expected it to set the fan mode to Balanced, but it randomly sets a fixed RPM.

**Desktop (please complete the following information):**
 - OS: archlinux, kernel 4.18

Example output of my status command:

```
Dev=0, CorsairLink Device Found: H100i V2!

Vendor: Corsair
Product: H100i V2
Firmware: 2.8.0.0
Temperature 0: 48.30 C
Temperature 1: 48.30 C
Temperature 2: 48.30 C
Fan 0:	Mode 0x07
	Current/Max Speed 1020/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 2940/0 RPM
```

audiohacked, 2018-08-18:
Be sure you are on the latest 'testing' branch and try again. It looks like you're using an older version of OpenCorsairLink.

farnoy, 2018-08-18:
It does seem a lot better, thanks! the read-outs of fan&pump modes, and fan RPM are not available. I get warnings like `Unsupported Generic Pump Function`, but it does change the settings reliably now.

closed 2018-08-18 by audiohacked