hyperkineticnerd/OpenCorsairLink #128: Hardware Testers List

audiohacked, 2018-08-19:
**Is your feature request related to a problem? Please describe.**
As author of OpenCorsairLink, I don't always have the hardware to test.

**Describe the solution you'd like**
I'd like to setup a list here of people who are willing to test OpenCorsairLink when there's a change to the software.

**Describe alternatives you've considered**
I have thought of finding developers who could/would be invested enough in OpenCorsairLink to take "ownership" of a particular driver like the RMi driver. And at least be able to test architectural changes against one device in a classification.

**Additional context**
N/A

aaroknight, 2018-09-16:
I use the H100i Pro Liquid Cooling System. The "Generic pump function" still seems to be unsupported in the latest feature release. If needed I will test and report.

buck2202, 2018-09-27:
`Corsair product detected. Checking if device is Lighting Node PRO... Dev=0, CorsairLink Device Found: Lighting Node PRO!`

You're probably aware of it already, but FWIW, [this](http://rh-software.com/) massive program seems to successfully control both LNP channels and multiple strips per channel (from virtualbox windows guest, ubuntu host) without the continuous flood of data I saw from iCUE on wireshark.

aaroknight, 2018-09-27:
> `Corsair product detected. Checking if device is Lighting Node PRO... Dev=0, CorsairLink Device Found: Lighting Node PRO!`
>
> You're probably aware of it already, but FWIW, [...]

Thank you very much. I didn't know this existed. But my goal is to run it directly from macOS without having a VM running. I just want to type in a command in Terminal to alter the colours. Like I do with my Pump RGB.

buck2202, 2018-09-27:
Agreed! I'd like that as well. Sorry, I didn't mean that to be addressed to you specifically; I just hoped it might be a help in reverse-engineering, depending on whether the devs have already done that part for the node pro.

In the meantime, though, one nice thing about SIV is that its settings seem to persist after exiting (unlike iCUE, which reverts to the 100% brightness rainbow thing when it exits)--it's still not as convenient as a CLI control, but you can at least configure the LEDs without having a virtual machine running continuously in the background.

TLable, 2019-04-23:
I'm running a Corsair H80i. I would like to change the LED to animate and change colors and also stay a certain color on startup. I can manually guess the coding to end up turning the LED off then to change as a new color back on.

audiohacked, 2019-04-25:
As a note to everyone: This is here to collect individuals who have said, "I have **this Corsair device** to test with". This thread is not for making suggestions on other software people can use, new features that need implemented, or any other conversations.

 Thank you,
Audiohacked

buck2202, 2019-04-25:
Sorry for the clutter. The relevant part of my first comment was that I have the Lighting Node PRO (LED strips only) for testing

`Corsair product detected. Checking if device is Lighting Node PRO... Dev=0, CorsairLink Device Found: Lighting Node PRO!`

JackPohlmann, 2019-04-29:
I'm running a Commander Pro and RGB LL120s. Let me know if there's anything I can help with.

paul-chambers, 2019-05-06:
I have a Corsair AX1200i I'd like to monitor. I am an experienced embedded Linux developer. The system is a Threadripper-based machine running ProxMox (i.e. Debian 9 underneath).

```
root@vm:~/OpenCorsairLink#` ./OpenCorsairLink.elf --dump
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (1d6b:0002)...
Checking USB device 2 (1d6b:0003)...
Checking USB device 3 (1d6b:0002)...
Checking USB device 4 (0781:5588)...
Checking USB device 5 (1d6b:0003)...
Checking USB device 6 (1d6b:0002)...
Checking USB device 7 (1d6b:0003)...
Checking USB device 8 (8087:0aa7)...
Checking USB device 9 (1b1c:1c00)...
Corsair product detected. Checking if device is USB Dongle... Dev=0, CorsairLink Device Found: USB Dongle!
Checking USB device 10 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = -1
```

lissyx, 2019-05-13:
I have a Corsair AX1600i and I'm willing to help.

definelicht, 2019-05-28:
I have an RM750i and cast test as necessary.

msmaldi, 2019-06-18:
Memory Corsair Vengeance, What do I use to control the RGB?

audiohacked, 2019-06-19:
@msmaldi It is unknown when Memory will be supported. Corsair RAM uses a totally different system (SMBUS) than what OpenCorsairLink uses (USB). Also, wrong thread to be posting in.

eitch, 2019-07-17:
I have a Obsidian 1000d, Commander Pro and a few Lightning Nodes to control my 16 fans and their RGB LEDs, so i can test that.

inkubux, 2019-09-16:
I have a H80i GT V2
I'm willing to help test if you need.

denangeles, 2019-11-11:
Can confirm this works with a Commander mini, I only have 3 pin fans, but setting fan mode of 2 (Default) and 5 (Performance) works. Cannot set RPM with mode 1.

Appreciate your work, this is amazing!

raw-bin, 2019-12-28:
Hi. I have a Corsair One i164 and am willing and able to help with testing and extending OCL.

TimZaman, 2020-01-26:
AX1600i here. Printing out any information seems non functional.
```
sudo  ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: AX1600i!
```

hgrahamcs, 2020-04-04:
I have a Corsair Commander Pro and I'm willing to help test.

janpython, 2020-04-08:
I've a Corsair H150i and your program seems to work fine with it! I'm able to adjust fan speed, pump LED and pump speed as far as I can oversee. Let me know if you nee more information!
$ sudo ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H150i Pro!

Vendor: Corsair
Product: H150i Pro
Firmware: 1.0.4.0
Temperature 0: 35.70 C
Fan 0:	Mode 0x00
	Current/Max Speed 1314/0 RPM
Fan 1:	Mode 0x00
	Current/Max Speed 1303/0 RPM
Fan 2:	Mode 0x00
	Current/Max Speed 1303/0 RPM
Pump:	Mode 0x01 (AsetekProPumpBalanced)
	Current/Max Speed 2040/0 RPM

luke2389, 2020-04-08:
> I've a Corsair H150i and your program seems to work fine with it! I'm able to adjust fan speed, pump LED and pump speed as far as I can oversee. Let me know if you nee more information!
> $ sudo [...]

I have the same one... so if you need another feedback, let me know.
Great job! :)

janpython, 2020-04-08:
> I have the same one... so if you need another feedback, let me know.
> Great job! :)

Ok, it's a quiet AIO-cooler so far (new build). But GNU/Linux support from Corsair is poor, unfortunately. So I'm trying to rewrite script which I found here: https://forum.level1techs.com/t/corsair-h100i-linux-howto/75829. It was written for an earlier version of OpenCorsairLink :-)
Did you manage to show temperature? With sudo ./OpenCorsairLink.elf --device 0 --led channel=0,mode=5 it's red, even at low temperatures...
Thanks for help!

erikostrom, 2020-04-17:
I have a Corsair H110i GTX that I'd be willing to test for:
```
e@bessie ~/g/g/OpenCorsairLink> sudo ./OpenCorsairLink.elf
Dev=0, CorsairLink Device Found: H110i GTX!

e@bessie ~/g/g/OpenCorsairLink>
```

chris17453, 2020-04-21:
AX1500i here, willing to dev. The corsair link code is in c# and a quick decompile will show you all you need from the hid module.

makk50, 2020-05-11:
I have an h115i rgb platinum and I and willing to help with testing. I'm no developer but I can provide you any info yo ask me.
Thank ypu very much for your effort

gardotd426, 2020-05-16:
@makk50 is it working with the H115i? There are like 20 different versions of the H115i and it seems like some of them work but some don't (or might not), and with seemingly no alternatives I'm struggling with which to buy. So far I've seen this many versions of the H115i for sale:

iCUE H115i RGB Pro XT (this is the one I wanted to buy, but I have no idea if it's supported and can't tell from the "Supported Devices" list if that's it or not.

CORSAIR HYDRO SERIES H115i

H115i PRO

H115i RGB PLATINUM, which is yours, I reckon.

Legion2, 2020-05-16:
The iCUE H115i RGB Pro XT is a very new device, released in 2020. It maybe only share the name with the older AIOs but works completely different on the driver side.

gardotd426, 2020-05-16:
@Legion2 yeah, I figured as much, I'm not too terribly concerned with RGB control or fan speed since I can just plug fans into the motherboard or my controller, but I want to make sure I have pump control because I'm not risking my 3800X, y'know? I saw that liquidctl is about to merge Kraken X*3 Gen compatibility and already has a fork working with pump control and monitoring and all that good stuff so I went ahead and got an X63.

makk50, 2020-05-16:
Not yet, they are still working on the protocol as far as I know.