hyperkineticnerd/OpenCorsairLink #129: Corsair H80i not detected

milmber, 2018-08-25:
**Describe the bug**
Corsair H80i not detected under MacOS

**To Reproduce**
Steps to reproduce the behavior:
1. Clone OpenCorsairLink repo
2. Install libusb using HomeBrew
3. Set CPATH to point to libusb.h
4. Build OpenCorsairLink
5. Check device is listed in lsusb
> Bus 020 Device 017: ID 1b1c:0c04 CORSAIR MEMORY INC. Integrated USB Bridge 
> Bus 020 Device 016: ID 1b1c:0c06 CORSAIR MEMORY INC. RM-Series C-Link Adapter 
> Bus 026 Device 001: ID 8087:8009 Intel Corporation Hub 
> Bus 029 Device 001: ID 8087:8001 Intel Corporation Hub 
> Bus 000 Device 001: ID 1d6b:PTHB Linux Foundation USB 3.0 Bus 
> Bus 000 Device 001: ID 1d6b:IPCI Linux Foundation USB 2.0 Bus 
> Bus 000 Device 001: ID 1d6b:IPCI Linux Foundation USB 2.0 Bus

6. Run OpenCorsairLink
7. No devices are detected

> ./OpenCorsairLink.elf --debug
> Checking USB device 0 (1b1c:0c04)...
> Corsair product detected. Checking if device is H80... 
> ---- Packet dump: -----------------------------
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> \-----------------------------------------------
> No (device_id 0x00)
> Corsair product detected. Checking if device is Cooling Node... 
> ---- Packet dump: -----------------------------
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> \-----------------------------------------------
> No (device_id 0x00)
> Corsair product detected. Checking if device is Lighting Node... 
> ---- Packet dump: -----------------------------
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> \-----------------------------------------------
> No (device_id 0x00)
> Corsair product detected. Checking if device is H100... 
> ---- Packet dump: -----------------------------
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> \-----------------------------------------------
> No (device_id 0x00)
> Corsair product detected. Checking if device is H80i... 
> ---- Packet dump: -----------------------------
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> \-----------------------------------------------
> No (device_id 0x00)
> Corsair product detected. Checking if device is H100i... 
> ---- Packet dump: -----------------------------
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> \-----------------------------------------------
> No (device_id 0x00)
> Corsair product detected. Checking if device is Whiptail... 
> ---- Packet dump: -----------------------------
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> \-----------------------------------------------
> No (device_id 0x00)
> Corsair product detected. Checking if device is H100i GT... 
> ---- Packet dump: -----------------------------
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> \-----------------------------------------------
> No (device_id 0x00)
> Corsair product detected. Checking if device is H110i GT... 
> ---- Packet dump: -----------------------------
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> \-----------------------------------------------
> No (device_id 0x00)
> Corsair product detected. Checking if device is H110i... 
> ---- Packet dump: -----------------------------
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> \-----------------------------------------------
> No (device_id 0x00)
> Checking USB device 1 (1b1c:0c06)...
> Checking USB device 2 (05ac:8508)...
> Checking USB device 3 (05ac:9226)...
> Checking USB device 4 (0a5c:21e8)...
> Checking USB device 5 (05ac:1105)...
> Checking USB device 6 (05ac:9126)...
> Checking USB device 7 (8087:8001)...
> Checking USB device 8 (8087:8009)...
> 
> DEBUG: scan done, start routines
> DEBUG: device_number = -1

**Expected behavior**
H80i should be detected

vlinkz, 2018-08-25:
@milmber, try using sudo/root

milmber, 2018-08-25:
@vlinkz - unfortunately I get the same result when running 

> sudo ./OpenCorsairLink.elf

llamafilm, 2018-08-31:
Any idea why it's showing as "Integrated USB Bridge" instead of the model name?  I have an H110i showing the same thing:

```
Integrated USB Bridge:

  Product ID:	0x0c04
  Vendor ID:	0x1b1c  (CORSAIR MEMORY INC.)
  Version:	2.00
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Corsair Memory, Inc.
  Location ID:	0x14600000 / 11
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0
```

JimBartlett, 2018-09-01:
I have the Lighting Node PRO, RM850i and the H110i running on macOS High Sierra 10.13.6 on a Gigabyte Z370i Gaming 7 Mobo. It's showing my H110i as a"Integrated USB Bridge" instead of the model name, same error I believe as @llamafilm

macOS is also not showing the RM850i name, but OpenCorsairLink is finding it regardless.

If I run Windows 10 under Parallels Desktop and pass my three Corsair devices, all are recognised and work with the Corsair software.

I can provide IORegistryExplorer files if you want of the USB, but would prefer to send directly as I the file requires the IORegistryExplorer macOS app, and there might be privacy issues if I attach it here. Hope you understand.

This file is the debug O/P from OpenCorsairLink
[OpenCorsairLink_macOS_debug.txt](https://github.com/audiohacked/OpenCorsairLink/files/2341853/OpenCorsairLink_macOS_debug.txt)

This file is the verbose output from lsusb (I installed lsusb using brew on macOS).
[OpenCorsairLink_lsusb_macos.txt](https://github.com/audiohacked/OpenCorsairLink/files/2341891/OpenCorsairLink_lsusb_macos.txt)

audiohacked, 2018-09-01:
If you look at the debug output `No (device_id 0x00)`, it shows the device id. Now, device_id of 0x00, is an invalid id. Either you don't have access to the Hardware, or Something's wrong and I can't fix it.

closed 2019-10-02 by audiohacked