hyperkineticnerd/OpenCorsairLink #131: Unsupported Generic Pump Function

botnarib, 2018-08-30:
I have a H100i V2 pump and am trying to set the pump mode. Unfortunately, I am getting the error "Unsupported Generic Pump Function" when setting the following commands to either set the fan or pump modes/speeds:

_sudo ./OpenCorsairLink.elf --device=0 --pump mode=5_

or

_sudo ./OpenCorsairLink.elf --device=0 --fan mode=3_

Also, it does not display the correct values for fan and pump RPMs. Here is what I get:

Dev=0, CorsairLink Device Found: H100i GT V2!

Vendor: Corsair
Product: H100i GT V2
Firmware: 2.8.0.0
Temperature 0: 30.00 C
Temperature 1: 30.00 C
Temperature 2: 30.00 C
Fan 0:	Mode 0x16
	Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
Pump:	Mode 0x05
	Current/Max Speed 1230/0 RPM
Setting pump to mode: 5
Unsupported Generic Pump Function

Your help is very much appreciated! Thank you!

audiohacked, 2018-08-31:
Well, you're giving the H100i an unsupported pump and fan mode.

closed 2018-09-01 by audiohacked

audiohacked, 2018-09-23:
@felixsanz Considering All I get is opened Issues from everyone and NO ONE opens a PULL REQUEST with a FIX. I'm not really encouraged to work or give a fix. I could take my code and take it down.

audiohacked, 2018-09-23:
THE WHOLE POINT OF MAKING THE CODE OPEN SOURCE IS SO I'M NOT FIXING EVERYTHING MYSELF!

audiohacked, 2018-09-23:
The thing is: you are not polite and the way you write is in a demanding voice.

felixsanz, 2018-09-23:
i'm just demanding that you could treat the users of your project a bit more nicely. it is OK if you have no time to fix things, but no need to be mad just because we ask for help. this is a library that is really important for us because no one wants an oven inside the PC case, so of course we are going to report issues. And if we could fix them we wouldn't be asking for help.

audiohacked, 2018-09-23:
Obviously you didn't bother to read the GPL License and I repeat it just the same in the header of my README.md. "It is licensed under the GNU General Public License (version 2) in the hope that it will be useful, but with NO WARRANTY of any kind." It would be nice if people take a day to learn C, it's not hard and I know you are a programmer.