hyperkineticnerd/OpenCorsairLink #134: H100i GTX - output only zeros

frasou, 2018-09-04:
Hello,
I have a **H100i  GTX**.

as root when I execute 
`./OpenCorsairLink.elf --device 0`

the output are only zeros:
>Dev=0, CorsairLink Device Found: H100i GTX!

>Vendor: Corsair
>Product: H100i GTX
>Firmware: 0.0.0.0
>Temperature 0:  0.00 C
>Temperature 1:  0.00 C
>Temperature 2:  0.00 C
>Fan 0:  Mode 0x00
>        Current/Max Speed 0/0 RPM
>Unsupported Generic Pump Function
>Pump:   Mode 0x00
>        Current/Max Speed 0/0 RPM

**When I try to change the pump mode I get this:**
`./OpenCorsairLink.elf --device 0 --pump mode=5`

>.
>.
>.
>Pump:   Mode 0x05
>       Current/Max Speed 0/0 RPM
>Setting pump to mode: 5
>Unsupported Generic Pump Function

**When I try to change LED colors, it works but output is same:**
`./OpenCorsairLink.elf --device 0 --led channel=1,mode=0,colors=00ff00`
>Dev=0, CorsairLink Device Found: H100i GTX!

>Vendor: Corsair
>Product: H100i GTX
>Firmware: 0.0.0.0
>Temperature 0:  0.00 C
>Temperature 1:  0.00 C
>Temperature 2:  0.00 C
>Fan 0:  Mode 0x00
>        Current/Max Speed 0/0 RPM
>Unsupported Generic Pump Function
>Pump:   Mode 0x00
>        Current/Max Speed 0/0 RPM

And I get green LED.

`lsusb -s 007 -v > /tmp/lsusb.txt 2>&1`
[lsusb.txt](https://github.com/audiohacked/OpenCorsairLink/files/2346817/lsusb.txt)

**Could you please help me with this issue?** My CPU temperatures raised significantly and I would appreciate reading from the cooler whether the pump is working properly. Thank you.

djurin, 2018-10-01:
Same for me...

dhanjit, 2018-11-05:
I am getting the same issue.

```
❯ sudo OpenCorsairLink --debug 
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (1d6b:0002)...
Checking USB device 2 (1d6b:0003)...
Checking USB device 3 (04d9:1203)...
Checking USB device 4 (1bcf:0002)...
Checking USB device 5 (1b1c:0c03)...
Corsair product detected. Checking if device is H100i GTX... Dev=0, CorsairLink Device Found: H100i GTX!
Checking USB device 6 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = -1
```

Now trying to set pump mode

```
❯ sudo OpenCorsairLink --device 0 --pump mode=5
Dev=0, CorsairLink Device Found: H100i GTX!

Vendor: Corsair
Product: H100i GTX
Firmware: 0.0.0.0
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Fan 0:	Mode 0x00
	Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
Pump:	Mode 0x05
	Current/Max Speed 0/0 RPM
Setting pump to mode: 5
Unsupported Generic Pump Function
```

I thought the `device=-1` was the issue. So, changing the device to -1 results in:

```
~ 7s
❯ sudo OpenCorsairLink --device -1 --pump mode=5
Dev=0, CorsairLink Device Found: H100i GTX!

~
```

definelicht, 2019-05-28:
Same issue. Used to work, now getting (with a fresh clone):

```
% sudo ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: RM750i!

Vendor: 
Product: 
Firmware: NA
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Powered: 0 (0d.  0h)
Uptime: 0 (0d.  0h)
Supply Voltage:  0.00 V
Total Watts:  0.00 W
Output 12v:
	Voltage  0.00 V
	Amps  0.00 A
	Watts  0.00 W
Output 5v:
	Voltage  0.00 V
	Amps  0.00 A
	Watts  0.00 W
Output 3.3v:
	Voltage  0.00 V
	Amps  0.00 A
	Watts  0.00 W
```

I've updated by Ubuntu packages since then, but otherwise nothing has changed in the system.

I can confirm that my Windows Virtual Box _does_ read values, so the issue must be on the Linux driver side.

definelicht, 2019-05-28:
Any adivce on how can I check if the issue is with OpenCorsairLink, libusb, or otherwise?

audiohacked, 2019-06-19:
@definelicht I just tested the 'testing' branch via CentOS 7 VM on Windows 10, the RM750i shows up just fine. So there has to be an issue with usb or libusb on your system.

[image]

lolzballs, 2019-06-25:
I'm having this issue also on Arch Linux (libusb 1.0.22). 

```
 ❯ ~ sudo opencorsairlink --device 0
Dev=0, CorsairLink Device Found: H100i GTX!

Vendor: Corsair
Product: H100i GTX
Firmware: 0.0.0.0
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Fan 0:	Mode 0x00
	Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
Pump:	Mode 0x00
	Current/Max Speed 0/0 RPM
```
This is the output libusb gives with `LIBUSB_DEBUG=4`: https://paste.hackendoz.com/VqA6cBQSXJ

Flashtekuk, 2019-06-26:
Similar issue of 0s when using --device=0

```$ ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H100i GT V2!

Vendor: Corsair
Product: H100i GT V2
Firmware: 0.0.0.0
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Fan 0:  Mode 0x00
        Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
Pump:   Mode 0x00
        Current/Max Speed 0/0 RPM
```

The output of `LIBUSB_DEBUG=4 ./OpenCorsairLink.elf --device 0 --debug ` command is available at https://gist.github.com/6a76eb30daefb6b3660e3ef8d6700221

Running on Gentoo with libusb v1.0.21

I hope this helps...

audiohacked, 2019-06-26:
@Flashtekuk @lolzballs The problem is due to your Linux not allowing OpenCorsairLink access to the hardware. I can't fix this.

lolzballs, 2019-06-27:
I'm running as root. Is there anything I can do to allow it to work?

Flashtekuk, 2019-06-27:
> @Flashtekuk @lolzballs The problem is due to your Linux not allowing OpenCorsairLink access to the hardware. I can't fix this.

So, even when running as root user my kernel is not allowing access to the USB device?

Is there anything you can think of that I could do to chase this down more, perhaps some kernel option or module to load up???

lolzballs, 2019-06-27:
I see `Unable to detach kernel driver from Corsair Device` in @Flashtekuk's log.

However, I don't have this, and verifying the kernel driver attached manually shows that no module is attached. (I've updated my original post with `--debug` flag).

gilvbp, 2019-07-07:
I have the same problem, How can i help?

tonylambiris, 2019-07-13:
Same issues here under Arch Linux using the latest code from testing:
```
➜ sudo OpenCorsairLink --device 0 --pump mode=5 --fan channel=0,mode=4
Dev=0, CorsairLink Device Found: H100i GT V2!

Vendor: Corsair
Product: H100i GT V2
Firmware: 0.0.0.0
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Fan 0:	Mode 0x00
	Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
Pump:	Mode 0x05
	Current/Max Speed 0/0 RPM
Setting fan to mode: 4
Setting pump to mode: 5
Unsupported Generic Pump Function
➜ pacman -Q | ag corsair
opencorsairlink-testing-git 20190624-1
```

I had to attach the cooler's USB header in VirtualBox for it to display correctly again:
[image]

After attaching to the official Windows 10 eval image (Corsair Link/iCue was installed on the virtual host):
```
➜ sudo pkg/opencorsairlink-testing/usr/bin/opencorsairlink --device 0 --debug
Checking USB device 0 (174c:55aa)...
Checking USB device 1 (2109:8110)...
Checking USB device 2 (04e8:61f5)...
Checking USB device 3 (2109:8110)...
Checking USB device 4 (1d6b:0003)...
Checking USB device 5 (046d:085c)...
Checking USB device 6 (0764:0501)...
Checking USB device 7 (1b1c:1b2e)...
Checking USB device 8 (1532:005c)...
Checking USB device 9 (2109:2811)...
Checking USB device 10 (1b1c:1b2d)...
Checking USB device 11 (2109:2811)...
Checking USB device 12 (18d1:5025)...
Checking USB device 13 (1d6b:0002)...
Checking USB device 14 (174c:3074)...
Checking USB device 15 (1d6b:0003)...
Checking USB device 16 (1038:1724)...
Checking USB device 17 (1038:1507)...
Checking USB device 18 (174c:2074)...
Checking USB device 19 (8087:0025)...
Checking USB device 20 (1b1c:0c09)...
Corsair product detected. Checking if device is H100i GT V2... Dev=0, CorsairLink Device Found: H100i GT V2!
Checking USB device 21 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H100i GT V2
Firmware: 2.8.0.0
22 03
Temperature 0: 34.30 C
22 03
Temperature 1: 34.30 C
22 03
Temperature 2: 34.30 C
E1
07 BC
Fan 0:	Mode 0xE1
	Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
0B 7C
Pump:	Mode 0x00
	Current/Max Speed 2940/0 RPM
DEBUG: deinit done
```

Hope that helps...

gilvbp, 2019-07-13:
@tonylambiris after doing your procedure, I'm getting this:
`
Corsair product detected. Checking if device is H115i Pro... Unable to detach kernel driver from Corsair Device
Unable to claim USB device interface
DEBUG: scan done, start routines
DEBUG: device_number = 1
Detected 0 device(s), submitted device 1 is out of range
`

aTosser, 2019-07-17:
There is an issue: I added:
`printf("Code: %i\n", rr);`
Under line 113 in logic/scan.c and the error code I got is -5 Which corresponds to LIBUSB_ERROR_NOT_FOUND. So either the scan is submitting the wrong device to libusb or the issue is libusb specific. I will play around with older versions of libusb and see what I can find on that end.

Gentoo, kernel-5.2.0, libusb-1.0.22, OCL-testing today.

MrPleasantEXE, 2019-07-18:
Just installed a new distro... guess im not the only one having this issue, any clues as to how to fix it?

Flashtekuk, 2019-07-23:
Latest code submission by @aTosser seems to make it play nice for me, good work :-D

aTosser, 2019-07-23:
That is not possible. Something else in your configuration must have changed. The line of code here in the comments just outputs the error code. I have made no other code submissions.

Flashtekuk, 2019-07-23:
Interesting...

$ ./OpenCorsairLink.elf --device 0 --debug
Checking USB device 0 (04c5:2028)...
Checking USB device 1 (1d6b:0003)...
Checking USB device 2 (0ac8:3420)...
Checking USB device 3 (046d:c05b)...
Checking USB device 4 (2334:000e)...
Checking USB device 5 (1d6b:0002)...
Checking USB device 6 (1d6b:0003)...
Checking USB device 7 (0408:3001)...
Checking USB device 8 (045e:00db)...
Checking USB device 9 (1d6b:0002)...
Checking USB device 10 (1d6b:0003)...
Checking USB device 11 (1d6b:0002)...
Checking USB device 12 (1d6b:0003)...
Checking USB device 13 (058f:6387)...
Checking USB device 14 (1b1c:0c09)...
Corsair product detected. Checking if device is H100i GT V2... Dev=0, CorsairLink Device Found: H100i GT V2!
Checking USB device 15 (0a12:0001)...
Checking USB device 16 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H100i GT V2
Firmware: 2.8.0.0
22 08
Temperature 0: 34.80 C
22 08
Temperature 1: 34.80 C
22 08
Temperature 2: 34.80 C
38
00 00
Fan 0:  Mode 0x38
        Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
06 EA
Pump:   Mode 0x00
        Current/Max Speed 1770/0 RPM
DEBUG: deinit done

aTosser, 2019-07-23:
Have you updated your system recently? Perhaps libusb or a kernel update fixed it.

lolzballs, 2019-07-27:
> `printf("Code: %i\n", rr);`
> Under line 113 in logic/scan.c and the error code I got is -5 Which corresponds to LIBUSB_ERROR_NOT_FOUND. So either the scan is submitting the wrong device to libusb or [...]

[libusb docs](http://libusb.sourceforge.net/api-1.0/group__dev.html#ga21bd343325f558987ca57e4e281a6d47) say that if `libusb_detach_kernel_driver` returns `LIBUSB_ERROR_NOT_FOUND`, it's because there was no kernel driver attached, so it's not an actual error.

okleinschmidt, 2019-08-10:
Same for me with macOS as reported in [191](https://github.com/audiohacked/OpenCorsairLink/issues/191)

audiohacked, 2019-08-13:
@okleinschmidt If you find out the difference between 'master' and 'testing', and find a fix. Please tell me, I've been getting the same complaint/"new issue" from a lot of people.

okleinschmidt, 2019-08-14:
> @okleinschmidt If you find out the difference between 'master' and 'testing', and find a fix. Please tell me, I've been getting the same complaint/"new issue" from a lot of people.

@audiohacked I would if I had C knowledge, but I unfortunately have no C knowledge. I was just making a guess, I don't know if it's right. But you should know where the difference is, you wrote the code?! ;-) btw., thanks for that! By now, I can't get it working either with master nor with testing branch. Nothing has changed to my OS, really strange.

audiohacked, 2019-08-14:
@okleinschmidt You should learn C, it's real easy! You can pop on to Freenode IRC and into `#opencorsairlink` and ask me any questions you have about the code.

lolzballs, 2019-08-14:
I just went through all the previous commits, starting from 30db298 (where the readme is updated to support h100i gtx), commits after 7c00ffc seem to not work on my system.

Not too sure if there's documentation anywhere about the asetek protocol so I'm not really sure where to go from here.

gilvbp, 2019-08-24:
@audiohacked how can I uninstall the opencorsair?

okleinschmidt, 2019-08-28:
> @gilvbp @audiohacked I am still seeing issues on the master branch

Yes, same for me.

```
$ sudo ./OpenCorsairLink.elf --device 0
Password:
Dev=0, CorsairLink Device Found: H115i Pro!

Vendor: Corsair
Product: H115i Pro
Firmware: 1.0.4.0
Temperature 0: 31.07 C
Fan 0:	Mode 0x00
	Current/Max Speed 660/0 RPM
Fan 1:	Mode 0x00
	Current/Max Speed 629/0 RPM
Unsupported Generic Pump Function
Pump:	Mode 0x00
	Current/Max Speed 2100/0 RPM
```

With master I can't control fan, pump RPM or anything else. With testing only with the "workaround" I mentioned earlier.

cheers,
Ole

jonasmalacofilho, 2019-08-30:
@audiohacked  

> If you find out the difference between 'master' and 'testing', and find a fix. Please tell me, I've been getting the same complaint/"new issue" from a lot of people.

The difference is fairly obvious: there used to be no check on the return value of `libusb_detach_kernel_driver` (see: [diff](https://github.com/audiohacked/OpenCorsairLink/compare/master...testing?expand=1&search=kernel#diff-26d8f2ce48bfcfe4fb47397957f31480))

There are two scenarios:

 - (a) There was no kernel driver bound to the device at the time of the call (for legitimate reasons).  This can happen, but I don't think it's common on normal Linux distros and kernel configs and without the user explicitly unbinding the driver.
 - (b) The libusb driver is already bound to the device, because the previous run of OpenCorsairLink (or, possibly, another software using libsub) has failed to (re) attach the (stock) kernel driver.

Either way, the simplest solution is probably to just check if a kernel driver is active before trying to detach it.¹  I can submit a patch in a few minutes but I'll only be able to roughly test it (don't have any of the supported devices).

---

¹ I couldn't spot any _obvious_ reason why OpenCorsairLink would fail to reattach the kernel driver when exiting, but I don't really know the codebase well enough.

okleinschmidt, 2019-09-03:
Great! Testing branch works for me again!

audiohacked, 2019-10-02:
Should be fixed with latest commit on `testing`.

closed 2019-10-02 by audiohacked