hyperkineticnerd/OpenCorsairLink #135: Lightning Node Pro support

zacjiang, 2018-09-12:
Would you be able to support lightning node pro? Thanks!

buck2202, 2018-09-27:
devs, what's current level of code support for this? I've gone deep enough to get [SIV](http://rh-software.com/) functioning in a windows virtualbox on ubuntu host, and it's able to control both channels and multiple strips per channel.

Unlike iCUE, its settings persist after program exit, and unlike iCUE, wireshark shows it changing led settings with limited bursts of communication rather a continuous flood. If SIV's communication methods are useful to you and you can sketch out what you need, I might be able to help.

audiohacked, 2018-09-28:
So, I use SIV to decipher what commands are for what function, but I use wireshark to capture all commands sent over USB. Currently, OCL should support all "known" commands for both Lightning Node Pro and Commander Pro. They share the same command set. What needs work is the "program" logic in `logic/settings/commanderpro.c`

audiohacked, 2018-09-28:
I'd like to also add is iCUE and keyboards sometimes aren't capable of "saving" settings. They don't have logic to run any of the effects found in iCUE.

eitch, 2019-07-17:
I have a LightningNodePro attached to my system, actually multiple of them to control the RGB of my 16 fans. OpenCorsairLink detects the LightningNodePro but when i select the device, it takes long to read it, but actually detects nothing, is this correct, or am i doing something wrong?

audiohacked, 2019-07-17:
@eitch Somethings going terribly wrong. I've seen that happen with my H115i Pro, under linux, and I have no idea how to work around that. The other CorsairLink devices work just fine, so it has to be something with the device.

commit-master, 2019-07-23:
Hey @audiohacked thanks for the work, I have a linux system with the Lightning Pro if you need testers I can help on that !

audiohacked, 2019-10-03:
I'm closing this issue until I can get the hardware myself. Until then I can't add support.

closed 2019-10-03 by audiohacked